goGPS-Project/goGPS_MATLAB #3: Implementation of the Vienna Mapping Function (VMF)

ege010, 2017-03-29:


syachroel, 2017-10-01:
After I get processing my rinex data one month, my next step just rechecks and compare Mapping function (Global mapping function, Niell mapping function, and Vienna mapping function) for the validation my result...
The question is, where I can look Mapping Function in goGPS...

Thanks

ege010, 2017-10-02:
Hi @syachroel 
at the moment goGPS applies only the GMF. The code for the Niell mapping function is already available (in the "goGPS\atmosphere" folder ), but it's not called during processing. The VMF is not implemented yet.
If you wish to try out the NiellMF code in PPP processing, you have to substitute the GMF here: https://github.com/goGPS-Project/goGPS_MATLAB/blob/master/goGPS/positioning/kalman/goGPS_KF_SA_code_phase_loop.m#L404
... and here: https://github.com/goGPS-Project/goGPS_MATLAB/blob/master/goGPS/atmosphere/tropo_error_correction.m#L153

ege010, 2017-10-03:
For future reference, on the addition of VMF to goGPS:
http://ggosatm.hg.tuwien.ac.at/DELAY/
https://link.springer.com/article/10.1007/s00190-017-1066-2

clicat, 2018-05-25:
VMF have been added to goGPS :-)

closed 2018-05-25 by clicat

syachroel, 2018-05-25:
Great.... thanks very much

syachroel, 2018-06-05:
Dear Dr. Gatti....

I Trie to execute Nightly version....but something miss...  :

>> goGPS
Error using Core.getInstance
Too many input arguments.

Error in goGPS (line 70)
    core = Core.getInstance(true); % Init Core

Thanks

clicat, 2018-06-05:
@syachroel 
Whenever you switch between branches of goGPS it is necessary to run the clearBranch script.
MATLAB retains the old Classes structure present in its cache, thus making necessary to force the cleaning. Pay also attention that the search path of MATLAB does not include paths from the old version and the new at the same time.

A small request for the future, please open a new issue for these kind of problems, it helps us to keep discussions on topic.

Let me know if it works

syachroel, 2018-06-05:
@DR.Gatti....

Ok,...already work and..I still learning new interface..

Thanks

commit dedaf6df (2018-11-23): phase idx bug fix #3