goGPS-Project/goGPS_MATLAB #8: Array comparison

CrawlerSpider, 2017-05-05:
In file IO_Settings.m, line 782 is a comparison between this.obs_type and this.REC_TARGET:
file_name = this.obs_full_name(this.obs_type == this.REC_TARGET);

where as i started goGPS the size of this.obs_type was 2 and the size of this.REC_TARGET was 1.
in addition this.obs_full_name was empty.

clicat, 2017-05-05:
Reviewing this issue I added a check on the size of obs_type and obs_name, but I did not understand the problem that you are reporting.
 
Let me try to explain how that part of the code should work:
In Class IO_Settings obs_name contains the list of the names of the files to be used for the various receivers.

**obs_name** it's a cell array containing in each position the name of the file to be processed, in the case that special keywords (such as ${DOY}, ${YY} and so on) are used obs_name is parsed and the keywords substituted. The result is then stored in a property called obs_full_name.

**obs_full_name** it's a cell of cell containing the real file name of the rinex to be processed, at the first level (the first index) I've the different receivers, for each of them I've the list of all their files.

_e.g.
`obs_name = {'MILA${DOY}${S}.${YY}o", "CATU${DOY}${S}.${YY}o"}`
from this and the other parameters of the session (e.g. sss_data_start) the following full name will be produced
`obs_full_name = {{'MILA3330.16o', 'MILA3340.16o'}, {'CATU3330.16o', 'CATU3340.16o'}}`
`obs_type = [0 1]`_

**obs_type** is an array that specify the function of each receiver

Looking at the line that you highlight
`file_name = this.obs_full_name(this.obs_type == this.REC_TARGET);`
is extracting the cell containing all the files relatives to the receivers marked as target
`this.obs_type` is an array 
`this.REC_TARGET` is a single value
`(this.obs_type == this.REC_TARGET)` return a logical array with the position of all the target receivers

To finish in the previous lines 
```
if isempty(this.obs_full_name)
    this.updateObsFileName();
end
```
`obs_full_name` is updated when empty.

I dunno if this is enough to solve the issue that you have encountered. Please tell me if you still have an issue or try to explain in a different way what is the problem that you found, so I can try to close the issue.
Thanks for the reporting!

CrawlerSpider, 2017-05-05:
The issue appears after kliking on "Import legacy inputs file" and then selecting the file settings.
Then I get the following error:

`Index exceeds matrix dimensions.`

`Error in IO_Settings/getTargetPath (line 782)
            file_name = this.obs_full_name(this.obs_type == this.REC_TARGET);
Error in IO_Settings/checkReceiverFiles (line 1692)
                    file_name_all = this.getTargetPath();
Error in IO_Settings/checkTargetFiles (line 1632)
            status = checkReceiverFiles(this, this.REC_TARGET, go_verbose);
Error in goGUIclass/updateLEDstate (line 2395)
                    status = this.state.checkTargetFiles();
Error in goGUIclass/checkUIdependencies (line 2200)
            this.updateLEDstate();
Error in goGUIclass/updateGUI (line 1561)
            this.checkUIdependencies();
Error in goGUIclass/browseINIFile (line 2567)
            this.updateGUI();
Error in goGUIclass/syncFromGUI (line 2330)
                this.browseINIFile();
Error in gui_goGPS>bINI_Callback (line 570)
    goGUI.syncFromGUI(goGUI.idUI.bINI);
Error in gui_mainfcn (line 95)
        feval(varargin{:});
Error in gui_goGPS (line 73)
    gui_mainfcn(gui_State, varargin{:});
Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)gui_goGPS('bINI_Callback',hObject,eventdata,guidata(hObject))
 Error using uiwait (line 81)
Error while evaluating UIControl Callback`

clicat, 2017-05-06:
I tried to investigate the problem, this probably happens importing an invalid ini file. Anyhow I updated the goGPS code with a bit more checks during and after the import of the settings. My latest update on the master branch (commit 51b5a80) should have fixed the issue.
Thanks again for the bug report, the code is now in fast evolution and so the number of bugs introduced recently is probably big. In the next weeks apart from improving a bit more the Meteo_Data class, we are going to work on bug-fixing, checking the main processing modes that we are currently using most (post-processing DD, PPP, SEID+PPP). 
Any feedback is always appreciated!

CrawlerSpider, 2017-05-06:
Problem is solved.

Thanks

closed 2017-05-06 by CrawlerSpider