goGPS-Project/goGPS_MATLAB #9: Losing precision when reading time values (e.g. from RINEX)

ege010, 2017-05-16:
A careful management of time values stored as double is needed already at data input (e.g. RINEX reading), to avoid introducing losing precision and/or introducing artificial time desynchronizations (see https://github.com/goGPS-Project/goGPS_MATLAB/commit/0d145145ffa79ab7406faebb92446a7ed3403da9#commitcomment-22147466).

clicat, 2017-05-17:
"Bug" fixed (hopefully) in rev. https://github.com/goGPS-Project/goGPS_MATLAB/commit/408ab5df0b9aab946121c0256f359d63552e59fa after some minor bug-fixes from rev. https://github.com/goGPS-Project/goGPS_MATLAB/commit/72e79382cee44935f3be4b76a1fd88da7a344a52 now the precision kept by load_RINEX should be improved. 
@lukegluke could you test whether or not your example works?

lukegluke, 2017-05-17:
Now there is swatooth everywhere :) (same rinex as in https://github.com/goGPS-Project/goGPS_MATLAB/commit/0d145145ffa79ab7406faebb92446a7ed3403da9#commitcomment-22147466)
[image]

Could you please write me you e-mail, so I can send you my rinex file.

clicat, 2017-05-17:
:-D very good @lukegluke here is my email: gatti . andrea at gmail . com

clicat, 2017-05-17:
New revision https://github.com/goGPS-Project/goGPS_MATLAB/commit/c00d86cabb6a5d334d45e58e35255686575adf74. I believe this solves the problem.
Times are still out of sync by some 0.1136 ps (1.136e-13 s).
pre-processing is not perfect -> it is changing the ph data, and it does have problems when diff time is zero but the phases are now smooth!

This precision problem is very annoying and difficult to handle...every time someone do an operation there is the risk of introducing new errors at the very small significant digits.

 @lukegluke thanks again for the report!

I'm closing the bug, but if there are still problems that I missed we can re-open it.

closed 2017-05-17 by clicat

lukegluke, 2017-05-18:
Seems to be fixed. Thanks!