goGPS-Project/goGPS_MATLAB #11: Typo

lukegluke, 2017-05-23:
multi_GNSS_biases_correction 68 line
k = find_eph(Eph, idx(c)+r, time_ref(round(length(time_ref))/2), 86400);

`time_ref(round(length(time_ref))/2)`
should be
`time_ref(round(length(time_ref)/2))`

clicat, 2017-05-23:
Bug corrected in rev https://github.com/goGPS-Project/goGPS_MATLAB/commit/1fa8391649de65f29642e6ff27a37864c15cafb6
Thanks again @lukegluke

clicat, 2017-05-23:
If you are interested in the project and want to contribute more directly we can provide you full access to the github repository. Anyhow we always appreciate suggestions and help in finding bugs.

closed 2017-05-23 by clicat

lukegluke, 2017-05-23:
Thanks for the offer, but I currently temporary use goGPS as a debug instrument, so it seems that soon I switch to other things. Can't guess how long would be this "soon", but for now, correcting bugs through opening issues is enough for me. Hope it's not so much inconvenience for you. Thanks for your work.

clicat, 2017-05-23:
That's absolutely not a problem, thanks for the help! Recently we are changing a lot of goGPS code and finding bugs is a very important task. We always appreciate who's able to point out defects that we can solve!