goGPS-Project/goGPS_MATLAB #15: Processing rover information bug

lukegluke, 2017-05-23:
With latest goGPS I have this:
[image]
instead of progress information:
[image]

Something change between c00d86cabb6a5d334d45e58e35255686575adf74 and 244be8bcfef49442587e33c80fdfa8993ab7bbad

clicat, 2017-05-24:
Position of the text fixed in rev: https://github.com/goGPS-Project/goGPS_MATLAB/commit/29b438b58d3279bb6a1099612a766cad0a42fc23

closed 2017-05-24 by clicat

lukegluke, 2017-05-24:
Confirm. Thanks.
[image]