goGPS-Project/goGPS_MATLAB #22: Meteorological RINEX file names are not using the ${...} keywords

ege010, 2017-09-05:
E.g. "ZIMM${DOY}0.${YY}m" does not work

clicat, 2017-09-05:
Solved in commit https://github.com/goGPS-Project/goGPS_MATLAB/commit/dee500f5ba31f51ae25e1f37f91632cd14413da0

syachroel, 2017-09-09:
Dear Dr Realini and Dr Gatti

Thank you for the answer...
I re-download, for update goGPS and 
processing its Ok for "ZIMM $ {DOY} 0. $ {YY} m", no problem ...
but I do not know there is a new problem with FTP?
as below ...

Computing Sun and Moon position... done
FTP connection to the AIUB server (ftp://ftp.unibe.ch). Please wait...
Error using ftp/mget (line 109)
FTP error: 425.

Error in download_dcb (line 144)
                mget(ftp_server,s1,down_dir);

Error in goGPS (line 533)
                            [file_dcb, compressed] = download_dcb([week_R(1) week_R(end)], [time_R(1) time_R(end)]);_

so I am not getting results in the OUT folder
Thank you,...

syachroel, 2017-09-11:
Dear Dr Realini and Dr Gatti

Thank you for the answer...
I re-download, for update goGPS and
processing its Ok for "ZIMM $ {DOY} 0. $ {YY} m", no problem ...
but I do not know there is a new problem with FTP?
as below ...

Computing Sun and Moon position... done
FTP connection to the AIUB server (ftp://ftp.unibe.ch). Please wait...
Error using ftp/mget (line 109)
FTP error: 425.

Error in download_dcb (line 144)
mget(ftp_server,s1,down_dir);

Error in goGPS (line 533)
[file_dcb, compressed] = download_dcb([week_R(1) week_R(end)], [time_R(1) time_R(end)]);_

so I am not getting results in the OUT folder
sorry ...one more... when I try in last version goGPS, I find more one times....looks as below...

Index exceeds matrix dimensions.

Error in IO_Settings/getMetFile (line 1114)
                    out = out{id};

Error in goGPS (line 3284)
                    md = Meteo_Data(state.getMetFile(session));

Thank you...

Regard

ege010, 2017-09-11:
Re-opening the issue here, although it's probably a different problem.

syachroel, 2017-09-11:
Ok...for temporary I try one by one with file meteorology...

Thanks

syachroel, 2017-09-16:
I try just for two files Obs,...but still like as below :

Index exceeds matrix dimensions.

Error in IO_Settings/getMetFile (line 1114)
out = out{id};

Error in goGPS (line 3284)
md = Meteo_Data(state.getMetFile(session));

wulingnuaa, 2017-09-19:
Dear syachroel , 
        I have the same problem with you as   "Computing Sun and Moon position... done
FTP connection to the AIUB server (ftp://ftp.unibe.ch). Please wait...
Error using ftp/mget (line 109)
FTP error: 425.

Error in download_dcb (line 144)
mget(ftp_server,s1,down_dir);

Error in goGPS (line 533)
[file_dcb, compressed] = download_dcb([week_R(1) week_R(end)], [time_R(1) time_R(end)]);_"

Do you solve it? 

 Then I have another problem with "ZIMM $ {DOY} 0. $ {YY} m in default_PPP/settings.ini, which is the update file in the web.   And how to solve the problem?   
Thank you a lot!

syachroel, 2017-09-19:
Hai.... wulingnuaa,

Actually, I want to get PWV for my RINEX data....
but just for one day, its mean one file*.obs and *.met, no problem...I get the result for PWV
so then I try for two files....unfortunately, the processing say: Index exceeds matrix dimensions
I don't know why so I just processing one by one and I compile for one-month data PWV

for FTP connection to the AIUB server (ftp://ftp.unibe.ch).
I think its little change for address, version goGPS before update

for "ZIMM $ {DOY} 0. $ {YY} m, for me its okay...
but in my case 

Error in IO_Settings/getMetFile (line 1114)
out = out{id};

Error in goGPS (line 3284)
md = Meteo_Data(state.getMetFile(session));

I think {getMetFile}, not yet finished program for file IO_Settings and goGPS

Regards

wulingnuaa, 2017-09-21:
Hi, syachroel

      Thank you very much for your reply!   I want to use goGPS as a plateform for my acadamic research on multi-frequency GPS/BDS real-time carrier phase ambiguity fixed. But up to now, I still can't use the goGPS software correctly. 

         Best regards!
          Your sincerely

2017-09-21
wuling@nuaa.edu.cn
 
From: syachroel
Date: 2017-09-19 13:27
To: goGPS-Project/goGPS_MATLAB
CC: wulingnuaa; Comment
Subject: Re: [goGPS-Project/goGPS_MATLAB] Meteorological RINEX file names are not using the ${...} keywords (#22)
Hai.... wulingnuaa,
Actually, I want to get PWV for my RINEX data....
but just for one day, its mean one file*.obs and *.met, no problem...I get the result for PWV
so then I try for two files....unfortunately, the processing say: Index exceeds matrix dimensions
I don't know why so I just processing one by one and I compile for one-month data PWV
for FTP connection to the AIUB server (ftp://ftp.unibe.ch).
I think its little change for address, version goGPS before update
for "ZIMM $ {DOY} 0. $ {YY} m, for me its okay...
but in my case
Error in IO_Settings/getMetFile (line 1114)
out = out{id};
Error in goGPS (line 3284)
md = Meteo_Data(state.getMetFile(session));
I think {getMetFile}, not yet finished program for file IO_Settings and goGPS
Regards

ege010, 2017-09-21:
Hi @syachroel and @wulingnuaa 

@clicat and I tried to fix your DCB FTP download issue (actually on our systems we did not get it), and we found a possible solution (https://github.com/goGPS-Project/goGPS_MATLAB/commit/888eac89860ec2d2458f8fd7ba136ad738d45d7c ). If the FTP download still causes you troubles, please download the needed DCB files manually from ftp://ftp.unibe.ch/CODE/

@syachroel we also tried to fix the problem relative to IO_Settings/getMetFile (https://github.com/goGPS-Project/goGPS_MATLAB/commit/677a8d6e8a6c7cf4aab90ff977b97ed5b514b170), please check and let us know if you find any issue!

wulingnuaa, 2017-09-21:
Hi, Eugenio Realini

     I also have problem with ftp download issue online. Could you help me to download some newly data for rover and master observition and navigation data for relative position with double differential.
thanks a lot.  
    Best regards

   Your sincerely!

2017-09-21
wuling@nuaa.edu.cn
 
From: Eugenio Realini
Date: 2017-09-21 16:09
To: goGPS-Project/goGPS_MATLAB
CC: wulingnuaa; Mention
Subject: Re: [goGPS-Project/goGPS_MATLAB] Meteorological RINEX file names are not using the ${...} keywords (#22)
Hi @syachroel and @wulingnuaa
@clicat and I tried to fix your DCB FTP download issue (actually on our systems we did not get it), and we found a possible solution (888eac8 ). If the FTP download still causes you troubles, please download the needed DCB files manually from ftp://ftp.unibe.ch/CODE/
@syachroel we also tried to fix the problem relative to IO_Settings/getMetFile (677a8d6), please check and let us know if you find any issue!

wulingnuaa, 2017-09-21:
Thank you a lot!

2017-09-21
wuling@nuaa.edu.cn
 
From: Eugenio Realini
Date: 2017-09-21 16:09
To: goGPS-Project/goGPS_MATLAB
CC: wulingnuaa; Mention
Subject: Re: [goGPS-Project/goGPS_MATLAB] Meteorological RINEX file names are not using the ${...} keywords (#22)
Hi @syachroel and @wulingnuaa
@clicat and I tried to fix your DCB FTP download issue (actually on our systems we did not get it), and we found a possible solution (888eac8 ). If the FTP download still causes you troubles, please download the needed DCB files manually from ftp://ftp.unibe.ch/CODE/
@syachroel we also tried to fix the problem relative to IO_Settings/getMetFile (677a8d6), please check and let us know if you find any issue!

syachroel, 2017-09-22:
Hi, Dr.Eugenio Realini

Thanks a lot,...
its done for execution my data for getting PWV for one week
right now still processing for one month....
Best regards

ege010, 2017-09-22:
@wulingnuaa please send me an email regarding the rover and master data for your tests.

Closing this issue since the problem is solved.

closed 2017-09-22 by ege010

syachroel, 2017-11-24:
Dear Dr. Realini

How are you, sorry..I disturb you again...

Anyway, I move to LINUX from WINDOWS for my processing data Rinex used go GPS

for getting PWV data, I was just thinking it is no problem with `go GPS`, but...I have some error messages like  below  :

Error using connect (line 22)

Invalid default value for property 'source' in class 'File_Wizard':

Could not open a connection to "igscb.jpl.nasa.gov", port "21".

Error in ftp (line 75)

connect(h)

Error in FTP_Downloader (line 104)

                this.ftp_server = ftp(strcat(this.addr, ':', this.port));

Error in Go_State/initProcessing (line 222)

            fw = File_Wizard;

Error in goGPS (line 154)

gs.initProcessing(); % Set up / download observations and navigational files

 

I just guess, the problem for download and connecting to ftp

But I don't know how to fix it

Thank you very much

for the answer