goGPS-Project/goGPS_MATLAB #23: Check PPP default project results (pre-processing???)

clicat, 2017-09-05:
The PWD results that the new code is now computing are smoother with respect to the "original" ones, the solution is more unstable at the beginning of the processing, and apparently the only difference that I've found is localized in the report. 
The old report is reporting some CS found in pre-processing, the new code does not find anything:

Cycle slips fixing                      #CS    Sat    Frq      Epoch      Float cycles     Fixed?       Fixed cycles
------------------------------------------------------------------------------------------------------
ZIMM2350.15O                             21    G02      1        105             0.254        NO       OBS. EXCLUDED
                                               G07      1       2509             0.197        NO       OBS. EXCLUDED
                                               G07      1       2558             0.058        NO       OBS. EXCLUDED
                                               G07      2       2509             0.153        NO       OBS. EXCLUDED
                                               G07      2       2558             0.045        NO       OBS. EXCLUDED
                                               G11      1       2424             0.172        NO       OBS. EXCLUDED
                                               G11      2       2424             0.134        NO       OBS. EXCLUDED
                                               G14      1       1108             0.366        NO       OBS. EXCLUDED
                                               G14      1       1114            -0.070        NO       OBS. EXCLUDED
                                               G15      1        765            -0.041        NO       OBS. EXCLUDED
                                               G15      1        779            -0.047        NO       OBS. EXCLUDED
                                               G17      1       2534             0.278        NO       OBS. EXCLUDED
                                               G22      1       1461             0.037        NO       OBS. EXCLUDED
                                               G22      1       2072             0.081        NO       OBS. EXCLUDED
                                               G22      1       2083             0.170        NO       OBS. EXCLUDED
                                               G22      2       1461             0.029        NO       OBS. EXCLUDED
                                               G22      2       2072             0.063        NO       OBS. EXCLUDED
                                               G22      2       2083             0.133        NO       OBS. EXCLUDED
                                               G23      1       2777             0.037        NO       OBS. EXCLUDED
                                               G23      2       2777             0.029        NO       OBS. EXCLUDED
                                               G28      1       2874             0.056        NO       OBS. EXCLUDED

I attach the old and new tropo results that I'm obtaining:
[zimm_PPP_023_settings.ini.txt](https://github.com/goGPS-Project/goGPS_MATLAB/files/1279005/zimm_PPP_023_settings.ini.txt)
[zimm_PPP_001_tropo.pdf](https://github.com/goGPS-Project/goGPS_MATLAB/files/1278992/zimm_PPP_001_tropo.pdf)
[zimm_PPP_001_tropo.txt](https://github.com/goGPS-Project/goGPS_MATLAB/files/1278993/zimm_PPP_001_tropo.txt)
[zimm_PPP_023_tropo.pdf](https://github.com/goGPS-Project/goGPS_MATLAB/files/1278995/zimm_PPP_023_tropo.pdf)
[zimm_PPP_023_tropo.txt](https://github.com/goGPS-Project/goGPS_MATLAB/files/1278996/zimm_PPP_023_tropo.txt)

ege010, 2017-09-06:
The reason for the smoothness of the new solution should be due to the relatively recent change of the sigmaq_tropo parameter default value. Which, by the way, is "hidden" because it's still located in goGPS.m instead of the ini file ;) BTW, also the default cutoff was set to 7 degrees in the default ini file, and the observation weighting was disabled (to improve the comparison with EPN official tropo products).

Anyway, I just pulled all recent changes and re-run default_PPP, getting the attached result. It's smoother than the older results (i.e. before sigmaq_tropo, cutoff and weighting changes), but it's not as smooth as your zimm_PPP_023_tropo.pdf. That's strange.

[zimm_PPP_001_tropo.pdf](https://github.com/goGPS-Project/goGPS_MATLAB/files/1280185/zimm_PPP_001_tropo.pdf)

ege010, 2017-10-02:
Found out the culprit: the settings.ini file for "default_PPP" had been changed, but probably it had not been pushed when I did the check above. Anyway, I restored the previous settings: 78913a5d264b8b341b0c0c0dae3b34181b8049b7

closed 2017-10-02 by ege010