goGPS-Project/goGPS_MATLAB #24: MATLAB version compatibility issue?

ege010, 2017-09-06:
From https://sourceforge.net/p/gogps/gogps-bugs/13/

Error using Go_State.getInstance (line 135)
Invalid default value for property 'cur_settings' in class 'Go_State':
Invalid default value for property 'UI_P_SRATE' in class 'Main_Settings':
The class goGUIclass has no Constant property or Static method named 'UI_P_SRATE'.

Error in goGPS (line 71)
gs = Go_State.getInstance();

>> ver

MATLAB Version: 9.2.0.556344 (R2017a)
MATLAB License Number: 593322
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 15063)
Java Version: Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode

MATLAB Version 9.2 (R2017a)
M_Map - mapping toolbox (Author: rich@eos.ubc.ca) Version 1.4h 1 Apr 
M_Map - mapping toolbox (Author: rich@eos.ubc.ca) Version 1.4h 1 Apr

clicat, 2017-09-08:
The error cannot be reproduced (I'm using MATLAB R2017a), the error sentence is also wrong:
"The class goGUIclass has no Constant property or Static method named 'UI_P_SRATE'."

the class HAS a constant named UI_P_SRATE
[image]

clicat, 2017-09-08:
@ege010 ...should we close the bug?

ege010, 2017-09-19:
Done

closed 2017-09-19 by ege010