goGPS-Project/goGPS_MATLAB #27: No check is present on the validity of the orbital files

clicat, 2017-09-19:
How to reproduce the error:
using a brdc of a day different from the one of the processing -> goGPS crash in pre-processing

syachroel, 2017-11-24:
Dear Dr. Realini/ Dr. Andrea

How are you, sorry..I disturb you again...

Anyway, I move to LINUX from WINDOWS for my processing data Rinex used go GPS

for getting PWV data, I was just thinking it is no problem with `go GPS`, but...I have some error messages like  below  :

Error using connect (line 22)

Invalid default value for property 'source' in class 'File_Wizard':

Could not open a connection to "igscb.jpl.nasa.gov", port "21".

Error in ftp (line 75)

connect(h)

Error in FTP_Downloader (line 104)

                this.ftp_server = ftp(strcat(this.addr, ':', this.port));

Error in Go_State/initProcessing (line 222)

            fw = File_Wizard;

Error in goGPS (line 154)

gs.initProcessing(); % Set up / download observations and navigational files

 

I just guess, the problem for download and connecting to ftp

But I don't know how to fix it

Thank you very much

for the answer

clicat, 2017-11-24:
Hello @syachroel the problem is due to igscb.jpl.nasa.gov beeing discontinued. In the latest updates of goGPS the connection to that server has been removed, and now it should work correctly.
Regarding PPP processing I'm working on it in the SEID_test branch that contains fixes and new parameters in the settings file to import the proper sigma values for the kalman filter. In particular now you have the possibility to change the following parameters:
#--------------------------------------------------------------------------------
[ATMOSPHERE]
#--------------------------------------------------------------------------------
# Standard deviation of a priori tropospheric delay (default = 0.200)
sigma0_tropo = 0.2
# Standard deviation of tropospheric delay [m/h] (default = 0.040)
std_tropo = 0.012
# Standard deviation of a priori tropospheric gradient (default = 1.000)
sigma0_tropo_gradient = 1
# Standard deviation of tropospheric gradient [m/h] (default = 0.010)
std_tropo_gradient = 0.005

clicat, 2017-11-24:
Now the SEID branch has been moved into master, update your version and the problem with the server should disappear!

syachroel, 2017-11-26:
Hello @clicat,

Thank you for the answer, and I already follow what your suggest,...
but I still found the message like above.....
Index exceeds matrix dimensions.

_**Error in RINEX_get_obs (line 76)
    sat_types_id(sat_types == 'S') = first_id_sys(7);

Error in load_RINEX_obs (line 199)
            [cur_line, obs] = RINEX_get_obs(file_buf, cur_line, num_sat, sat, sat_types, obsColumns, nObsTypes, cc, active_sys, first_id_sys);

Error in goGPS (line 458)
                        load_RINEX_obs(filename_obs, state.getConstellationCollector(), processing_interval);**_

Event with ZIMM or my data.
Thank you 

Regards
Syachroel

clicat, 2017-11-27:
@syachroel, you are totally right. In my local copy I'm testing various pieces of code and I've many uncommitted changes that made the code work, I updated the file that caused the problem: load_RINEX_obs (commit f291f56f078f37e2143134be02b7c40d5cab875d).
I also updated the settings file of the default PPP project to show some valid values for the new tropospheric parameters added in the latest updates.

syachroel, 2017-11-27:
Dear @cliclat
Alright, I will try again....

Thanks a lot

syachroel, 2017-11-27:
Dear @clicat,
thank you for your respons...

and the result...

**Computing Sun and Moon position... done
FTP connection to the AIUB server (ftp://ftp.unibe.ch). Please wait...
Error using ftp/mget (line 109)
FTP error: 425.

Error in download_dcb (line 144)
                mget(ftp_server,s1,down_dir);

Error in goGPS (line 577)
                            [file_dcb, compressed] = download_dcb([week_R(1) week_R(end)], [time_R(1) time_R(end)]);**

syachroel, 2017-11-28:
wait,...

Today
I try on another PC, its worked for ZIMM station
its make me confused,...and I come back to the PC one, it's not working...
still...any message like below

Error using ftp/mget (line 109)
FTP error: 425.

Error in download_dcb (line 144)
mget(ftp_server,s1,down_dir);

Error in goGPS (line 577)
[file_dcb, compressed] = download_dcb([week_R(1) week_R(end)], [time_R(1) time_R(end)]);**

thanks

clicat, 2017-11-28:
I think that your computer (or your MATLAB) appears to have some network problems in accessing ftp addresses, you receive  FTP error 425, "Can't open data connection."
You can probably download all the needed files manually and then run goGPS, or try to find out why your configuration has a problem with ftp addresses.

syachroel, 2017-11-28:
Ok thanks, I use another PC,..it's working very well
Thanks a lot

clicat, 2020-01-08:
This issue is relative to the old goGPS version, since 1.0 is coming I’m closing this bug!

closed 2020-01-08 by clicat