goGPS-Project/goGPS_MATLAB #29: GLONASS positioning problem

arglas864, 2017-09-22:
It seems GLONASS-only positioning doesn't work properly. If i try to post-process a RINEX file containing data of a car on a circuit, the positions of each lap slowly drift with time, and there are jumps when new satellites are acquired. I think this problem also affects GPS+GLONASS positioning.

syachroel, 2017-12-14:
Dear ege010 and clicat,
Btw, I just want to try for processing Glonnas data, but the result....

Pre-processing rover...
                 
  0% [                                                      ]
Subscript indices must either be real positive integers or logicals.

Error in interpolate_SP3_clock (line 64)
b = SP3_time(p) - time;

Error in transmission_time (line 107)
    dtS = interpolate_SP3_clock(time_tx_RAW, SP3, sat);

Error in satellite_positions (line 93)
    [time_tx(i,1), dtS(i,1)] = transmission_time(time_rx, pseudorange(i), sat(i), Eph(:,k), SP3, sbas, err_tropo(i), err_iono(i), dtR, frequencies,
    obs_comb, lambda(i,:));

Error in init_positioning (line 164)
    [XS, dtS, XS_tx, VS_tx, time_tx, no_eph, eclipsed, sys] = satellite_positions(time_rx, pseudorange, sat0, Eph, SP3, sbas, err_tropo, err_iono,
    dtR, frequencies, obs_comb, lambda, p_rate);

Error in pre_processing (line 294)
                        [XR_tmp, dtR_tmp, sat_pos_tmp, ~, ~, sat_vel_tmp, ~, ~, err_iono_tmp, sat, el_tmp, az_tmp, ~, ~, cov_XR_tmp, var_dtR_tmp, ~,
                        ~, ~, cond_num_tmp, bad_sat_i, bad_epochs(i), var_SPP(i,:), ~, eclipsed_tmp, ISBs_tmp, var_ISBs_tmp, y0, b, A, Q] =
                        init_positioning(time(i) - dt_pr(i), alpha1(sat0).*pr1(sat0,i) - alpha2(sat0).*pr2(sat0,i), snr1(sat0,i), Eph_t, SP3,
                        zeros(8,1), sbas_t, XR0, [], [], sat0, [], zeros(length(sat0),2), cutoff, snr_threshold, frequencies, p_rate, flag_XR, 0, 0,
                        nisbs > 1, 1); %#ok<ASGLU>

Error in goGPS (line 733)
                        [time_R_diff(:,1,f), pr1_R(:,:,f), ph1_R(:,:,f), pr2_R(:,:,f), ph2_R(:,:,f), pos_R_new(:,:,f), dtR(:,1,f), dtRdot(:,1,f),
                        el_r(:,:,f), az_r(:,:,f), bad_sats_R(:,1,f), bad_epochs_R(:,1,f), var_dtR(:,1,f), var_SPP_R(:,:,f), status_obs_R(:,:,f),
                        status_cs, eclipsed, ISBs, var_ISBs] = pre_processing(time_GPS_diff, time_R_diff(:,1,f), pos_R, pr1_R(:,:,f), ph1_R(:,:,f),
                        pr2_R(:,:,f), ph2_R(:,:,f), dop1_R(:,:,f), dop2_R(:,:,f), snr1_R(:,:,f), Eph, SP3, iono, lambda, frequencies, obs_comb,
                        nSatTot, w_bar, flag_XR, sbas, cc, flag_full_prepro, order);

Thanks for the respon

clicat, 2018-12-13:
Since we are working on the new static oriented version of goGPS the older KF code is no more maintained. In the future, we are going to re-add KF functionalities re-writing everything from scratch. I'm closing the bug for this reason

closed 2018-12-13 by clicat