goGPS-Project/goGPS_MATLAB #32: Index exceeds matrix dimensions.

niklgjer, 2017-11-05:
While attempting to run a least squares algorithm with code and phase double differences, an error occurrred when the program was writing the output file:

Index exceeds matrix dimensions.

Error in goGPS (line 3406)
                                        fprintf(fid_out, row_str, date_R(i,1), date_R(i,2), date_R(i,3),
                                        date_R(i,4), date_R(i,5), date_R(i,6), week_R(i), tow(i), phi_KAL(i),
                                        lam_KAL(i), h_KAL(i), X_KAL(i), Y_KAL(i), Z_KAL(i), NORTH_UTM(i),
                                        EAST_UTM(i), h_ortho(i), utm_zone(i,:), nsat(i), HDOP(i), KHDOP(i),
                                        NORTH_KAL(i), EAST_KAL(i), UP_KAL(i), fixed_amb(i), succ_rate(i),
                                        estim_tropo(i), ZWD(i), PWV(i));

clicat, 2017-11-24:
Hi @niklgjer sorry for the delay, I did not look at the git page since a while ago... Unfortunately I cannot reproduce the error that you are getting, could you maybe share with me the project folder that you are using?

clicat, 2018-01-30:
I'm temporary closing this issue since it is not possible to reproduce the bug.

closed 2018-01-30 by clicat