goGPS-Project/goGPS_MATLAB #35: Prolbem with goGPS_BLK_DD_code_phase_static.m

wulingnuaa, 2018-01-30:
Dear Eugenio
        I have down loaded the newly version of goGPS of 0.5.2. Is there any problem with multi-epoch (batch) least-squares adjustment? I feel the program is imcomplete.

clicat, 2018-01-30:
Yes it is, we were trying to develop a "block" solution, but at a certain point of the development we found out that to complete that approach we needed to improve some other components of the sw (pre-processing), and the only way to do it in a "simple" way was to rewrite all the code, so the development has been suspended. We started developing the "Nightly" branch for this reason. Up to now we just implemented a rough "Block" PPP solution, we are trying to stabilize it and make it works with multiple constellations. When most of the modules we need will be coded we are going to implement other kinds of solutions (DD, Kalman Filter processing...). 
To test block DD solution, we are currently using a branch called "**block_version_paper**" that contains an older version of the block approach, it is very poorly maintained and not aligned with the Master branch, everything runs fine only if all the cycle slips are found by the pre-processing.
The idea for the development of the goGPS software is to spend less time on the Master, and focus all our attention to the "Nightly" branch, it will be the only place were block solution will be developed. In the future, we are planning to stabilize that version, create a new interface that works with it and fix as many bugs as we can.

clicat, 2018-04-13:
Closing because the request is referred to an experimental feature that will be developed and rewritten only for the new version

closed 2018-04-13 by clicat