goGPS-Project/goGPS_MATLAB #38: Problem starting the Matlab GUI v 0.4.3

mohamedamjad, 2018-04-13:
When I tried to open the GUI I got:
```
Error using exist
The first input to exist must be a string scalar or character vector.

Error in goGUIclass/updateLEDstate (line 2168)
                                        if exist([data_path file_name],'file')

Error in goGUIclass/checkUIdependencies (line 1914)
            obj.updateLEDstate();

Error in goGUIclass/updateGUI (line 1369)
            obj.checkUIdependencies();

Error in goGUIclass/syncFromGUI (line 2021)
                    obj.updateGUI();

Error in goGUIclass/importStateMatlab (line 2662)
            obj.syncFromGUI(obj.idUI.lProcMode);

Error in goGUIclass/initInterface (line 1312)
                obj.importStateMatlab([obj.getSettingsDir() obj.defaultSettingsFile]);

Error in goGUIclass/init (line 246)
            obj.initInterface();

Error in goGUIclass (line 195)
            obj.init(handles);

Error in gui_goGPS_unix_linux>gui_goGPS_unix_OpeningFcn (line 78)
goGUI = goGUIclass(handles, goGUIclass.isUnix);

Error in gui_mainfcn (line 220)
    feval(gui_State.gui_OpeningFcn, gui_hFigure, [], guidata(gui_hFigure), varargin{:});

Error in gui_goGPS_unix_linux (line 62)
    [varargout{1:nargout}] = gui_mainfcn(gui_State, varargin{:});

Error in goGPS (line 118)
        [mode, mode_vinc, mode_data, mode_ref, flag_ms_pos, flag_ms, flag_ge, flag_cov, flag_NTRIP, flag_amb, ...
 
```

mohamedamjad, 2018-04-13:
with the actual version in branch master I got this error:
```
Error using Go_State.getInstance (line 135)
Invalid default value for property 'cur_settings' in class 'Go_State':
Invalid default value for property 'UI_P_SRATE' in class 'Main_Settings':
The class goGUIclass has no Constant property or Static method named 'UI_P_SRATE'.

Error in Go_State.getCurrentSettings (line 154)
                this = Go_State.getInstance();

Error in Core (line 61)
            this.state = Go_State.getCurrentSettings();

Error in Core.getInstance (line 71)
                this = Core();

Error in goGPS (line 67)
core = Core.getInstance();
 
```

Is there a working version of gogps? If yes, how gan I download it

clicat, 2018-04-13:
The error you are receiving is probably caused by some variables or state that are coming from another branch of goGPS.
Try running the script "clearBranch" before relaunching goGPS and it should work!
Let me know if this solved your problem!

clicat, 2018-04-16:
It does: https://github.com/goGPS-Project/goGPS_MATLAB/blob/master/goGPS/utility/clearBranch.m

It just performs a cleaning of all the singleton classes, typically needed when switching branch.
Restarting MATLAB has the same effect.
Sometimes is useful to delete the last_settings.ini file!

Anyhow, I'm happy that you manage to run it: I'm closing the issue.

closed 2018-04-16 by clicat

clicat, 2018-04-16:
You are right the version from gogps-project.org is older than the one in the master branch, we will try to fix this in the near future. Thanks

LakshmiCBR, 2020-10-03:
goGPS GUI initialization completed in 30.70 seconds
Error using exist
The first input to exist must be a string scalar or character vector.

Error in goGUIclass/updateLEDstate (line 2168)
                                        if exist([data_path file_name],'file')

Error in goGUIclass/checkUIdependencies (line 1914)
            obj.updateLEDstate();

Error in goGUIclass/updateGUI (line 1369)
            obj.checkUIdependencies();

Error in goGUIclass/browseINIFile (line 2338)
            obj.updateGUI();

Error in goGUIclass/syncFromGUI (line 2035)
                obj.browseINIFile();

Error in gui_goGPS>bINI_Callback (line 359)
    goGUI.syncFromGUI(goGUI.idUI.bINI);

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in gui_goGPS (line 64)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)gui_goGPS('bINI_Callback',hObject,eventdata,guidata(hObject))
 
Error using uiwait (line 81)
Error while evaluating UIControl Callback.

Please helpme running goGPS.....it is failing with the above errors. I tried using clearBranch.m too. But no use