goGPS-Project/goGPS_MATLAB #50: goGPS vs. Alpha2 Nightly

syachroel, 2018-06-06:
Dear Dr. Gatti

I have tried Nightly version and running well but first I receive the message before running well, and a second after finished computation, I check the workstation the data result in its empty also in folder output.
For temporary that's all, thank you for attention

Regards
Syachroel

clicat, 2018-06-06:
mmm... it's strange that it is saying that the configuration is invalid, I do not see anything wrong (just warnings), but that's normal, the software should create those directories.
For the output, well the new nightly version works in a different way, it produces objects that can be "interrogated" at the end of the processing.

after a successful lunch of goGPS you should run getResults, and in the workspace, you will be able to see the objects rec/rec_list/core
for example by typing rec_list.showZTD you should be able to see the ZTD as computed
(if you executed PPP)

I'm checking the testing function of the configuration in these days, so keep the version updated and in a few days from now re-try to run your project.

Please, attach your config file so I'll be able to understand better what went wrong.

syachroel, 2018-06-06:
I choose to file ZIMM then the config, I select one by one from PPP setting and setting nightly,...
I think there's default right, and running processing until...getResult, but I check...the workstation still empty, for example, ZTD: NaaN...and HDOP...empty.
...
So I tried with my data, the result it's same...empty

Thanks

syachroel, 2018-06-25:
great.....thanks Dr.Gatti

I tried today for the last Version goGPS, v 0.6.0 alpha 2 - nightly ...for data ZIMM
The result ....Successfully for temporary, for the next, I will tray for another data

Regard

Syachrul

closed 2018-06-25 by syachroel