goGPS-Project/goGPS_MATLAB #53: Rinex File Unable to run Completely

zain-farooq, 2018-09-21:
Hi!
I am using Huace i80 RTK set for getting master and rover data. The data collected is for around 16 minutes. goGPS runs for quite some time before displaying the error. 
"
Subscript indices must either be real positive integers or logicals.

Error in goGPS (line 3075)
                    time_GPS_diff(unused_epochs) = [];
"
Is it that 16 minutes data is large for the buffers, since I checked 'yamatogawa' files and they span 13 minutes. (The trajectory displayed is correct, as this is how we collected data and also processed on RTKlib)

[Rinex Data.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/2404970/Rinex.Data.zip)

clicat, 2018-12-13:
Dear @zain-farooq sorry if no one answered you before. We are working on a new version of goGPS for static positioning and at the moment the Kalman filter approach for dynamic studies hasn't been re-implemented yet. Unfortunately looking at the old code could be very time demanding for us, there could be a bug but we have no time to fix it. We are spending all of our time developing the newer goGPS version. Thank you very much for your feedback and for sending us useful data that we will use for testing. If I'll find the time to look at the older goGPS version I'll try your dataset!
Best regards
Andrea Gatti

clicat, 2019-02-17:
Dynamic positioning is not yet implemented, closing the "bug" temporary

closed 2019-02-17 by clicat