goGPS-Project/goGPS_MATLAB #55: Processing Rate on goGPS 1.0 Beta

syachroel, 2018-11-22:
Dear Dr. Gatti

Have a nice time in Bali,...
I am already trying to validate my RINEX data for **one day** with IGS data (ZPD). 
In goGPS 1.0 Beta, I have good correlation and diff average. But, if I want to **more one-day** data
with PROCESSING RATE EVERY 5 MINUTES, how to change from 30 seconds to 5 minutes??

Thanks a lot

Syachrul Arief

Best Regard

clicat, 2018-11-22:
Actually the ZPD solution from IGS is averaged so you can average the solution you get from gogps in a span of 5 minute. Note that to compute a comparable value you should use the same mapping function used by the comparison data and maybe play a bit with the regularisation parameters of the tropospheric delay. @giuliot   tested the solution against various official products, maybe he can tell to you how to calibrate the processing parameter to match the igs output

clicat, 2018-11-22:
Ehm it’s not @giuliot but @tagliagiulio

syachroel, 2018-11-23:
Thanks for the answer,...
But how to insert average the solution to goGPS 1.0 Beta @tagliagiulio
in goGPS master any optional for 5 minutes...

giulio-t, 2018-11-25:
Dear syachroel, there is no option to have one parameter every 5 minute. If you want to compare the solutions against Igs official zpd, you will have to average them in a 5 minute span.
I am writing the class Tropo_Sinex_Compare to do that, but is still in development

clicat, 2018-12-13:
In the actual version, we added spline solution (linear or cubic) you can compute it a linear spline with an interval of 5 minutes, and then undersample the results taking one value every 5 minutes