goGPS-Project/goGPS_MATLAB #58: File (*.clk) on products data CDDIS and GARNER in 1999

syachroel, 2018-12-21:
Dear Dr. Andrea Gatti,

I process rinex 1999 data with goGPS, and the result:

  [WW] WARNING: igs09744.clk_30s have not been found remotely
  [WW] WARNING: igs09744.clk_30s have not been found remotely
  [WW] WARNING: igs09744.clk have not been found remotely
  [WW] WARNING: igs09744.clk have not been found remotely
         Some files not found remotely
 
  _[!! ] ERROR: Not all ephemerids found program might misbehave files
Error using File_Wizard / conjureNavFiles (line 736)
With no ephemerides, the processing cannot be completed_

then, I tried crosscheck to the web:
http://garner.ucsd.edu/pub/products/0974

I found that there are not files **igs09744.clk**, but what I get is the file **ig109744.clk**

the question is how to change the code so that the file can still be accessed

Thanks a lot

Syachrul Arief

clicat, 2018-12-23:
Actually, I see that igs09744.sp3.Z is available on the CDDIS server, I believe that the downloader of goGPS was not able to go online and for some unknown reason, it didn't download the needed files. You can try to launch goGPS again or download manually the required orbits/clocks and put them into the folder that you have chosen on the Resources tab of the interface

[image]

clicat, 2019-02-15:
Hi @syachroel, I hope that with the latest update from yesterday the download of ephemerides is more stable! Let me know and close the bug if you think is solved.

syachroel, 2019-02-15:
Okay... thanks for information..
I will try

syachroel, 2019-02-15:
Dear @clicat,

I just tried ... for the data 2009 it's running well, but my 1999 data ... at below on the report

--------------------------------------------------------------------------
 Starting session 1 of 2

 Begin 1999/01/07 00:00:00.0000000 GPS
 End   1999/01/07 23:59:59.9999000 GPS
--------------------------------------------------------------------------
 [ -- ] Preparing receiver 1 of 1

   **   Importing ephemerides...
 [ WW ] Warning: No ephemerides have been found at D:\goGPS_MATLAB-goGPS_1.0_beta2\data\satellite\EPH\0991\igs09912.sp3
 [ WW ] Warning: No ephemerides have been found at D:\goGPS_MATLAB-goGPS_1.0_beta2\data\satellite\EPH\0991\igs09913.sp3
 [ WW ] Warning: No ephemerides have been found at D:\goGPS_MATLAB-goGPS_1.0_beta2\data\satellite\EPH\0991\igs09914.sp3
 [ WW ] Warning: No ephemerides have been found at D:\goGPS_MATLAB-goGPS_1.0_beta2\data\satellite\EPH\0991\igs09915.sp3
 [ WW ] Warning: No ephemerides have been found at D:\goGPS_MATLAB-goGPS_1.0_beta2\data\satellite\EPH\0991\igs09916.sp3
   **   Loading antennas phase center variations
        Opening file I14.ATX for reading

**_Struct contents reference from a non-struct array object.

Error in Core_Utils.readAntennaPCV (line 354)
                                                        if ~(date_limits.first.getMatlabTime() >
                                                        datenum(validity_start) &&
                                                        (date_limits.last.getMatlabTime() <
                                                        datenum(validity_end)))

Error in Core_Sky/loadAntPCV (line 1950)
            this.ant_pcv = Core_Utils.readAntennaPCV(filename_pcv, this.cc.getAntennaId(),
            this.time_ref_coord);

Error in Core_Sky/initSession (line 179)
                this.loadAntPCV(this.state.getAtxFile);

Error in Core/initSkySession (line 804)
                this.sky.initSession(time_lim.first, time_lim.last);

Error in Core/prepareSession (line 778)
                        this.initSkySession(time_lim_large);

Error in Core/go (line 841)
                            is_empty = this.prepareSession(s);

Error in goGPS (line 137)
            core.go(); % execute all_**

Thanks

clicat, 2019-02-16:
Ok Ok, I see that I was reading sp3 while you were writing clk! My mistake I did not get your problem before! Yes, at that time there were no clk products from IGS. I cannot actually find the "ig109744.clk" that you are talking about. Anyhow, I'll try to process some old data (never tried before) and correct the problems that the software may find! 
Thanks and sorry for the misunderstanding, that was totally on my part!

clicat, 2019-02-17:
There was a millennium bug :-D in the GPS_Time object and some other minor issue.
I also added the new center "Legacy" that is basically an IGS GPS repo that accepts the fact that there could be an sp3 without external clocks.
By the way, you can create new "centers" in that file using the combination of resources that you prefer. 

Unfortunately, I was working on a test set and I notice that prior to 1 May 2000 there's a problem in computing good orbits (related to the sp3 with no external clocks) thus producing a bad code positioning and then an over flagging of almost all the epochs. We will try to correct the problem during the week. I never tried to process "old" station files before!

I'll keep you updated as soon as I discover if the problem with old data is solved!

clicat, 2019-02-17:
Of course, now I remember May 2000 is the limit for the selected availability! Before that date, goGPS is not able to work properly! Totally different outlier detection is needed for those data, the actual function flags everything as outlier! I close the "bug" and I'll open an enhancement request (but I definitely don't know when we are going to try to process data older than 2000!).
Thanks for your help that made me discover some other bugs!

Reference:
[https://www.gps.gov/systems/gps/modernization/sa/data/](https://www.gps.gov/systems/gps/modernization/sa/data/)

closed 2019-02-17 by clicat

syachroel, 2019-02-18:
Great.....thanks for help my answer....