goGPS-Project/goGPS_MATLAB #59: Pre Processsing error at CRX file

JKeast0, 2019-02-13:
goGPS doesn't create any file in C:\...\satellite\CRX directory. And I'm getting this error message. How to do I? Is it possible to deactivate pre-processing because of my data not regularly?

`  **   Pre-processing on receiver 1: log12040207a
  --------------------------------------------------------------------

   **   Removing observations marked as bad in Bernese .CRX file
   **   Reading CRX data
 [ WW ] Warning: The required (updated) CRX files were not found in C:\goGPS_MATLAB-goGPS_1.0_beta\data\satellite\CRX directory.
                 

   **   Removing observations for which no ephemerid or clock is present
Dot indexing is not supported for variables of this type.

Error in  -  (line 1695)
            [unix_time2, unix_time_f2] = gt_2.getUnixTime();

Error in Receiver_Work_Space/remBad (line 1030)
            coord_ref_time_diff = first_epoch - this.sat.cs.time_ref_coord;

Error in Receiver_Work_Space/preProcessing (line 7702)
                this.remBad();

Error in Command_Interpreter/runPrePro (line 882)
                        rec(r).work.preProcessing();

Error in Command_Interpreter/exec (line 713)
                            this.runPrePro(rec, tok(2:end));

Error in Core/exec (line 883)
            this.cmd.exec(this.rec, cmd_list, level);

Error in Core/go (line 845)
                            this.exec(cmd_list(execution_block == eb), sss_level(execution_block == eb));

Error in goGPS (line 137)
            core.go(); % execute all`

clicat, 2019-02-14:
Thanks for the report, I'll try to reproduce the error and correct it in the morning!

JKeast0, 2019-02-14:
But my data isn't IGS point, and can you give me a link, please?

clicat, 2019-02-14:
I reviewed the download functions, I was not really able to reproduce the bug, but I modified and corrected some procedures. Give the new code a try!
The CRX file contains a list of satellite problems, and they can be found here: ftp://ftp.aiub.unibe.ch/BSWUSER52/GEN/ one file per year
e.g. ftp://ftp.aiub.unibe.ch/BSWUSER52/GEN/SAT_2019.CRX

However, I do not understand what's the meaning of the "IGS" sentence, why an IGS point should be special?

If you find the bug again send me the complete log, maybe I can understand a little better if the problem was generated due to some previous misconfiguration.

And thanks again for trying to use goGPS, I know that with no documentation (not yet ready) it can be hard!

JKeast0, 2019-02-14:
Thank you so much. I didn't know what is meaning ".crx" file.

closed 2019-02-14 by JKeast0