goGPS-Project/goGPS_MATLAB #60: Error at Downloading Resourcues

JKeast0, 2019-02-14:
**When I selected the "Allow automatic download of missing resources" MATLAB gives me this error message. Actually, my problem is the auto download. I had downloaded resources manually at the previously I submitted the issue. Detailed log file in here at my google drive account
Thanks for your attention**

[https://drive.google.com/file/d/1Q9MaoBdHw4qX44_OEB5AKPkgjzambbn_/view?usp=sharing](url)

> 
 
 [ WW ] WARNING: brdm0390.19p have not been found locally
 [ WW ] WARNING: brdm0380.19p have not been found locally
 [ WW ] WARNING: brdm0370.19p have not been found locally
        Some files not found locally
 
        Checking remote folders ...
 
 [ ok ] brdm0390.19p found (on remote server cddis)
 [ ok ] brdm0380.19p found (on remote server cddis)
 [ ok ] brdm0370.19p found (on remote server cddis)
        All files have been found remotely
 
        Downloading Resources ...
 
Error using fwrite
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in Core_Utils.aria2cDownloadUncompress (line 619)
                            fwrite(fid, str, '*char');

Error in File_Wizard/navigateTree (line 311)
                        f_status_lst = Core_Utils.aria2cDownloadUncompress(file_name_lst, f_ext_lst, f_status_lst,
                        date_list);

Error in File_Wizard/conjureResource (line 249)
                        [status, ~] = this.navigateTree(file_tree, 'download');

Error in File_Wizard/conjureNavFiles (line 767)
                status = this.conjureResource(list_preferred{i}, date_start, date_stop);

Error in File_Wizard/conjureFiles (line 517)
                this.conjureNavFiles(dsa, dso);

Error in Core/prepareProcessing (line 666)
            fw.conjureFiles(time_lim_large.first, time_lim_large.last);

Error in goGPS (line 131)
            core.prepareProcessing();
>

clicat, 2019-02-15:
Hi @JKeast0 try to pull the latest updates from github from yesterday! I suppose that this problem was generated by the fact that no folder "reserved" was present in the goGPS directory. This caused the impossibility of creating the temporary file needed by the aria2 execution to download files. Now if not present I create it. In an update from yesterday I reviewed the download procedure, and in fact, in my version, I have the "fwrite(fid, str, '*char');" command on a different line!

clicat, 2019-04-29:
I'm closing this issue since no more feedback has been provided to check that the problem is still there (it shouldn't be)

closed 2019-04-29 by clicat