goGPS-Project/goGPS_MATLAB #62: Matlab 2018b with Linux

syachroel, 2019-03-08:


clicat, 2019-03-09:
Hi, I’m on holiday and far from home till Sunday. On Monday I’ll have a look at it.
1. There is a missing check on the existence of the antex file
2. It seems that the antenna haven’t been loaded: this could due to 2 possible problems, the first is that I broke the code (very probable with my last rushed update on the parallelism), the second is that the antex (.ATX) file doesn't exist (in a case sensitive way.

As I said, on Monday I’ll debut the code ;-)

syachroel, 2019-03-09:
Sorry....,okay I'm waiting on Monday

clicat, 2019-03-19:
Hi @syachroel sorry if I didn't answer I was too busy, I tried to reproduce the bug with no success, could you attach the antex (.ATX) file you were trying to load?

clicat, 2019-08-16:
with no updates since March, I close this bug

closed 2019-08-17 by clicat

kazimz, 2020-11-19:
Hi Andrea,
I am facing exactly the same issue. Kindly have a look. Thanks
ATX file attached. 
[I14.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/5569567/I14.zip)
Best Regards
Ali