goGPS-Project/goGPS_MATLAB #63: Error at downloading files from 'ascp01950.436' to 'ascp02050.436'

gdpinchina, 2019-03-28:
It seems urls `ftp://ssd.jpl.nasa.gov/*` are deprecated. Without files from 'ascp01950.436' to 'ascp02050.436', the Core_Sky.m cannot create `de436.bin`. Any solutions to this?

clicat, 2019-03-28:
🤔 It's working for me: ftp://ssd.jpl.nasa.gov/pub/eph/planets/ascii/de436/
I can attach it for you, you can put it into "goGPS_MATLAB/goGPS/reserved/JPL/" if the folder do not exist you can create it
[ascp02050.436.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/3018357/ascp02050.436.zip)

clicat, 2019-03-28:
@gdpinchina if you need additional files I have a copy of that entire folder on my hard drive

gdpinchina, 2019-03-29:
Thanks a lot! I really appreciate that you upload the files 😄 . 
I have used VPN to access the website, but it still didn't work 🤔 . Maybe the Great wall in China blocked it in some other ways.  
By the way, I notice that NASA seems to moving its ftp resources to its https website. As it shows the announce on `https://ssd.jpl.nasa.gov/`: 
```
IMPORTANT: The SSD Announcement email list is now available. Major changes are coming in the near future, including possible hostname and/or URL changes. To be notified of such changes, subscribe to this email list.
```
and on `https://podaac.jpl.nasa.gov/`:
```
PO.DAAC FTP services will be retired on 3 June 2019. For more information and alternate methods of access, please see our latest Announcement.
```

closed 2019-03-29 by gdpinchina

gdpinchina, 2019-03-29:
I think I need two more files, or maybe everything in your folder `goGPS_MATLAB/goGPS/reserved/JPL/`? I'm not sure.
Here are these two files:
` ftp://ssd.jpl.nasa.gov/pub/eph/planets/ascii/de436/header.436`
`ascp01950.436`
Again, thanks a lot!

clicat, 2019-03-29:
Thanks a lot for all the info.
Hi @gdpinchina, thanks!
Here I attach all the necessary files (if I discover that they can be redistributed in the goGPS repository I'll copy them in it, today I'm going to investigate this).
Anyway, we will switch the protocol of JPL resources from FTP to https thanks to your info! @giulio-t or me will do it!

[ascp01950.436.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/3021637/ascp01950.436.zip)
[header.436_177.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/3021638/header.436_177.zip)
[header.436.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/3021639/header.436.zip)

gdpinchina, 2019-04-01:
Thank you, my friends!