goGPS-Project/goGPS_MATLAB #66: Cannot run goGPS.m under Matlab

libing64, 2019-08-12:
>> goGPS

               ___ ___ ___
     __ _ ___ / __| _ | __|
    / _` / _ \ (_ |  _|__ \
    \__, \___/\___|_| |___/
    |___/                    v 1.0 beta 2

--------------------------------------------------------------------------
 [ WW ] Warning: This is a goGPS beta build, use it at your own risk!
                 Some parts of it could not work properly
                 Please open a new issue on github if you found any bug
--------------------------------------------------------------------------

   **   Building settings object...

 [ ok ] File "last_settings.ini" found, settings imported!
        Opening file G:\source\gnss\goGPS_MATLAB\data\project\default_PPP\station\CRD\stations.crd for reading
 [ ok ] goGPS GUI initialization completed in 8.09 seconds
 
 Checking ZIMM
 Checking ZIM2
 Checking ZIMJ
 File availability checked
 [ WW ] Warning: The settings field flag_separate_coo_at_boundary is not valid => using default 0
 [ ok ] File "G:\source\gnss\goGPS_MATLAB\data\project\default_PPP\config\settings.ini" found, settings imported!
        Opening file G:\source\gnss\goGPS_MATLAB\data\project\default_PPP\station\CRD\stations.crd for reading
   **   Starting computation!

 Logging to: out\log\goGPS_run_20190812_232557.log
 
 Checking input files and folders...

 [ ok ] Home dir is present
 [ ok ] Observation dir is present
   **   Checking RINEX input files
        - ZIMM  (4 ok - 0 ko)
        - ZIM2  (4 ok - 0 ko)
        - ZIMJ  (4 ok - 0 ko)
 [ ok ] Observation rinex are present
 [ ok ] Coordinate dir is present
 [ ok ] Meteorological dir is present
        - zim2  (4 ok - 0 ko)
        - zimj  (4 ok - 0 ko)
        - zimm  (4 ok - 0 ko)
 [ ok ] Met rinex are present
 [ ok ] Ocean loading dir is present
 [ WW ] Warning: Atmospheric loading dir is missing (1)
 [ ok ] Antenna dir is present
 [ ok ] International Geomagnetic reference dir is present
 [ ok ] Ephemerides dir is present
 [ ok ] Clock Offset dir is present
 [ WW ] Warning: Earth Rotation Parameters dir is missing (1)
 [ ok ] Satellite Manouvers dir is present
 [ ok ] Differential Code Biases dir is present
 [ ok ] EGNOS Message Center dir is present
 [ ok ] Geoid file is present
 [ ok ] The Geoid is loaded
 [ WW ] Warning: Ionospheric map dir is missing (1)
 [ ok ] International Geomagnetic Reference Frame dir is present
 [ WW ] Warning: Vienna Mapping Function dir is missing (1)

   **   PROJECT: Default PPP project
       Satellite system "GPS" active
        - using frequencies: "L1, L2" 
       Satellite system "GLONASS" is inactive
       Satellite system "Galileo" is inactive
       Satellite system "QZSS" is inactive
       Satellite system "BeiDou" is inactive
       Satellite system "IRNSS" is inactive
       Satellite system "SBAS" is inactive
      
 Conjuring all the auxilliary files...
   **   Checking ephemerides / clocks / ERPs
        Checking local folders ...
 [ ok ] Ephemerides files are present ^_^
   **   Checking DCB files
 [ ok ] DCB files are present ^_^
   **   Checking CRX (Satellite problems file)
 [ ok ] CRX files are present ^_^
   **   Checking ionospheric resources files
        Checking local folders ...
 [ ok ] Ionosphere resource files are present ^_^
   **   Checking Athmospheric loading files
        Checking local folders ...
 [ ok ] Atmospheric loading files are present ^_^
   **   Checking VMF files
        Checking local folders ...
 [ ok ] Vienna Mapping Function files are present ^_^
        Opening file G:\source\gnss\goGPS_MATLAB\data\project\default_PPP\station\CRD\stations.crd for reading

   **   Executing: FOR S*
----------------------------------------------------------------------------

--------------------------------------------------------------------------
 Starting session 1 of 2

 Begin 2017/04/08 00:00:00.0000000 GPS
 End   2017/04/08 23:59:59.9999000 GPS
--------------------------------------------------------------------------
 [ -- ] Preparing receiver 1 of 3
 [ -- ] Preparing receiver 2 of 3
 [ -- ] Preparing receiver 3 of 3

   **   Importing ephemerides...
        Opening file igs19434.sp3 for reading
        Opening file igs19435.sp3 for reading
        Opening file igs19436.sp3 for reading
        Opening file igs19440.sp3 for reading
        Opening file igs19441.sp3 for reading
   **   Importing satellite clock files...
        Opening file igs19434.clk_30s for reading
        Opening file igs19435.clk_30s for reading
        Opening file igs19436.clk_30s for reading
        Opening file igs19440.clk_30s for reading
        Opening file igs19441.clk_30s for reading
   **   Loading antennas phase center variations
        Opening file I14.ATX for reading
   **   Sat Ephemerids: switching to antenna phase center
Index exceeds matrix dimensions.

Error in jplephem>state (line 404)
if (pjd(1) + pjd(4) < ss(1) || pjd(1) + pjd(4) > ss(2))

Error in jplephem (line 222)
[pv, rrd] = state(et2, list);

Error in Core_Sky/computeSunMoonPos (line 1919)
                rrd = jplephem(jd_tdb(e), sun_id, earth_id);

Error in Core_Sky/tabulateSunMoonPos (line 1955)
            [this.X_sun , this.X_moon] = this.computeSunMoonPos(this.getCoordTime());

Error in Core_Sky/sunMoonInterpolate (line 1793)
                this.tabulateSunMoonPos();

Error in Core_Sky/getSatFixFrame (line 1244)
            X_sun = this.sunMoonInterpolate(t_sun, true);

Error in Core_Sky/COMtoAPC (line 1354)
            [i, j, k] = this.getSatFixFrame(this.getCoordTime());

Error in Core_Sky/toAPC (line 1318)
                this.COMtoAPC(1);

Error in Core_Sky/initSession (line 187)
                    this.toAPC();

Error in Core/initSkySession (line 884)
                this.sky.initSession(time_lim.first, time_lim.last);

Error in Core/prepareSession (line 858)
                        this.initSkySession(time_lim_large);

Error in Command_Interpreter/exec (line 728)
                                    is_empty = core.prepareSession(s);

Error in Core/exec (line 920)
            this.cmd.exec(this, cmd_list);

Error in Core/go (line 905)
            this.exec(this.state.cmd_list);

Error in goGPS (line 137)
            core.go(); % execute all

libing64, 2019-08-12:
[goGPS_run_20190812_232557.log](https://github.com/goGPS-Project/goGPS_MATLAB/files/3493244/goGPS_run_20190812_232557.log)

libing64, 2019-08-12:
Matlab version R2017a

clicat, 2019-08-12:
Hi @libing64 as far as I see you had a crash in computing the celestial position of sun and moon. 
The function jplephem have not been written by us but it is used to read the jpl planetary ephemeris files stored in the file "de436.bin".
If this file is corrupted this function does not work. 
You can find the file and delete it, depending on the platform it could be in:
 - Windows - HOME_DIR\AppData\Local\goGPS 
 - LINUX - HOME_DIR/.goGPS
 - MacOS - HOME_DIR/Library/Application Support/goGPS
goGPS should have a correct copy of it in data/reference/JPL and it will automatically copy it to the final destination during execution.

Anyhow, I see that you are running goGPS beta 2, please download the latest version in a way that I can exclude old bugs that have been corrected. My suggestion is to keep the version updated by using git (git pull command).

WARNING:  these days we are all on holidays, and maybe we can be a little slow in answering questions ;-)

libing64, 2019-08-13:
@clicat Great thanks for your reply, I will have a try and I will also keep the version update.

libing64, 2019-08-13:
@clicat After updating to the latest code, the GUI seems not runnable.

Here is the output

>> goGPS

               ___ ___ ___
     __ _ ___ / __| _ | __|
    / _` / _ \ (_ |  _|__ \
    \__, \___/\___|_| |___/
    |___/                    v 1.0 beta 3jp

--------------------------------------------------------------------------
 [ WW ] Warning: This is a goGPS beta build, use it at your own risk!
                 Some parts of it could not work properly
                 Please open a new issue on github if you found any bug
--------------------------------------------------------------------------

   **   Building settings object...

 [ WW ] Warning: The settings field flag_free_net_tropo is not valid => using default 0
 [ WW ] Warning: The settings field tropo_spatial_reg_sigma is not valid => using default [ 1e-07 ]
 [ WW ] Warning: The settings field tropo_spatial_reg_d_distance is not valid => using default [ 25000 ]
 [ WW ] Warning: The settings field tropo_gradient_spatial_reg_sigma is not valid => using default [ 1e-07 ]
 [ WW ] Warning: The settings field tropo_gradient_spatial_reg_d_distance is not valid => using default [ 25000 ]
 [ ok ] File "last_settings.ini" found, settings imported!
        Opening file G:\source\gnss\goGPS_MATLAB\data\project\default_PPP\station\CRD\stations.crd for reading
No public property Tooltip exists for class matlab.ui.control.UIControl.

Error in GUI_Main/insertTabProcessing (line 616)
            grd.Children(end).Tooltip = 'This is not kept in case of snooping';

Error in GUI_Main/openGUI (line 239)
            this.insertTabProcessing(tab_panel);

Error in GUI_Main (line 96)
            this.openGUI();

Error in GUI_Main.getInstance (line 106)
                this = GUI_Main();

Error in Core_UI/openGUI (line 668)
            this.main = GUI_Main.getInstance();

Error in goGPS (line 105)
        ui.openGUI();
 
        Checking receivers data directories

libing64, 2019-08-14:
@clicat After resetting some commits to this position, the example would have another error.

commit 163c6b7f6cabaa8db1e99ea2f265e5cc7ba294d5 (HEAD -> goGps_2)
Author: Andrea <gatti.andrea@gmail.com>
Date:   Thu Jul 25 16:55:02 2019 +0200
    Moving note position in help

=========================================
goGPS

               ___ ___ ___
     __ _ ___ / __| _ | __|
    / _` / _ \ (_ |  _|__ \
    \__, \___/\___|_| |___/
    |___/                    v 1.0 beta 3jp

--------------------------------------------------------------------------
 [ WW ] Warning: This is a goGPS beta build, use it at your own risk!
                 Some parts of it could not work properly
                 Please open a new issue on github if you found any bug
--------------------------------------------------------------------------

   **   Building settings object...

 [ WW ] Warning: The settings field flag_free_net_tropo is not valid => using default 0
 [ WW ] Warning: The settings field tropo_spatial_reg_sigma is not valid => using default [ 1e-07 ]
 [ WW ] Warning: The settings field tropo_spatial_reg_d_distance is not valid => using default [ 25000 ]
 [ WW ] Warning: The settings field tropo_gradient_spatial_reg_sigma is not valid => using default [ 1e-07 ]
 [ WW ] Warning: The settings field tropo_gradient_spatial_reg_d_distance is not valid => using default [ 25000 ]
 [ ok ] File "last_settings.ini" found, settings imported!
        Opening file G:\source\gnss\goGPS_MATLAB\data\project\default_PPP\station\CRD\stations.crd for reading
        Checking receivers data directories
 [ ok ] goGPS GUI initialization completed in 3.18 seconds
 
   **   Starting computation!

 Logging to: out\log\goGPS_run_20190814_000142.log
 
 Checking input files and folders...

 [ ok ] Home dir is present
 [ ok ] Observation dir is present
   **   Checking RINEX input files
        - ZIMM  (4 ok - 0 ko)
        - ZIM2  (4 ok - 0 ko)
        - ZIMJ  (4 ok - 0 ko)
 [ ok ] Observation rinex are present
 [ ok ] Coordinate dir is present
 [ ok ] Meteorological dir is present
        - zim2  (4 ok - 0 ko)
        - zimj  (4 ok - 0 ko)
        - zimm  (4 ok - 0 ko)
 [ ok ] Met rinex are present
 [ ok ] Ocean loading dir is present
 [ WW ] Warning: Atmospheric loading dir is missing (1)
 [ ok ] Antenna dir is present
 [ ok ] International Geomagnetic reference dir is present
 [ ok ] Ephemerides dir is present
 [ ok ] Clock Offset dir is present
 [ WW ] Warning: Earth Rotation Parameters dir is missing (1)
 [ ok ] Satellite Manouvers dir is present
 [ ok ] Differential Code Biases dir is present
 [ ok ] EGNOS Message Center dir is present
 [ ok ] Geoid file is present
 [ ok ] The Geoid is loaded
 [ WW ] Warning: Ionospheric map dir is missing (1)
 [ ok ] International Geomagnetic Reference Frame dir is present
 [ WW ] Warning: Vienna Mapping Function dir is missing (1)

   **   PROJECT: Default PPP project
[... 146 lines cut ...]
Error using cat
Dimensions of matrices being concatenated are not consistent.

Error in cell2mat (line 75)
            m{n} = cat(2,c{n,:});

Error in Atmosphere/initVMF (line 290)
            h_data = cell2mat(textscan(fid,formatSpec,91));

Error in Core/prepareSession (line 925)
                            this.atmo.initVMF(time_lim_large.first,time_lim_large.last);

Error in Command_Interpreter/exec (line 788)
                                    is_empty = core.prepareSession(s);

Error in Core/exec (line 998)
            this.cmd.exec(this, cmd_list);

Error in Core/go (line 983)
            this.exec(this.state.cmd_list);

Error in goGPS (line 144)
            core.go(); % execute all
 
>>

clicat, 2019-08-14:
The VMF reader is not very robust it supposes that the files have always the same size. One probable cause of this error is that the downloaded VMF file is corrupted or it does not have the right size. check if they have all the same dimension, if not, delete the files, they will be re-downloaded, or try to make the processing without using VMF.

Let me know if it works!

libing64, 2019-08-14:
@clicat The vmf files all have the same size, but after deleting these files and re-downloading, they both have different size.

I tried to debug this code, and I found that it didn't stopped when reading VMF file, but when reading ~/data/reference/VMF/orography_ell.

[image]

clicat, 2019-08-14:
This is very strange: all my files have the same dimensions: 668'636 bytes (672 KB on disk)
And the dimension of orography_ell and orography are 106'986 bytes (111 KB on disk)
My suggestion is to delete that files and make a new pull from git. After making the pull, type "clearBranch" on MATLAB (or close it and reopen it), then try again.
These are files provided with the code, I see no reason why they cannot be loaded with the code, sorry.

closed 2019-12-17 by clicat