goGPS-Project/goGPS_MATLAB #71: NET Processing Error

JKeast0, 2019-11-19:
Hi the developer team of goGPS.  I've been trying to perform a NET process and it giving me the following errors..
On pre-processing:
**error in state- epoch out of range
** [ WW ] Warning: Code solution have not been computed, something is wrong in the current dataset
I tried with different datasets. It didn't work. Then in executing NET step says this:
 [ WW ] Warning: No reference have been found, using the mean of the receiver for the computation
 [ !! ] Error: Not enough receivers (< 2), skipping network solution
Also I want to use GOPE station as target and the others as reference. Maybe i'm putting some commands wrong. Here is the log file. 
[goGPS_run_20191119_144147.log](https://github.com/goGPS-Project/goGPS_MATLAB/files/3863585/goGPS_run_20191119_144147.log)

clicat, 2019-11-19:
Hello! First of all, let me say sorry because we did not yet produce any documentation yet, second because sometimes we change part of code and we risk to broke some other parts of goGPS (still beta).

Looking fastly to your question I can spot a first problem (maybe not the only one).

  cmd_001 = "FOR S*"
  cmd_002 = "FOR T4"
  cmd_003 = "LOAD T4 @30s"
  cmd_004 = "PREPRO T4"
  cmd_005 = "END"
  cmd_006 = "NET T4 R1:3 -IONO"
  cmd_007 = "PUSHOUT T4"
  cmd_008 = "END"
  cmd_009 = "SHOW T* ENUBSL"

You are trying to perform a network solution without loading T1 to T3 and you are using the NET command in the wrong way, it should be: "NET T1:4 R1:3 --IONO".
SHOW T* ENUBSL cannot actually show anything because only one value of the baseline is estimated in a session of one day.
We tried to run your test downloading the stations you are using and on our internal version, everything works as of now (14:40 UTC), please update the code.

This is the right command list you should use:
  cmd_001 = "FOR S*"
  cmd_002 = "FOR T*"
  cmd_003 = "LOAD T$ @30s"
  cmd_004 = "PREPRO T$"
  cmd_005 = "END"
  cmd_006 = "NET T1:4 R1:3 --IONO"
  cmd_007 = "PUSHOUT T*"
  cmd_008 = "END"

clicat, 2019-11-19:
The sequence of command is how it is written in the settings file, here there is the one you can paste into the GUI:

FOR S*
   FOR T*
      LOAD T$ @30s
      PREPRO T$
   END
   NET T1:4 R1:3 --IONO
   PUSHOUT T*
END
SHOW T* MAP

JKeast0, 2019-11-20:
Thanks for replying but I've tried with latest version and the commands you gave but still giving same errors. 

**   Pre-processing on receiver 1: CPAR
  --------------------------------------------------------------------
   **   Removing observations marked as bad in Bernese .CRX file
   **   Reading CRX data
        - SAT_2019.CRX [ ok ] 
 [ WW ] Warning:  2880 bad epochs (sat G04) have been discovered into CRX

   **   Removing observations for which no ephemerid or clock is present
 [ WW ] Warning: Satellite G18 have some missing clocks values
                 removing invalid epochs
 [ WW ] Warning: Satellites with missing clocks: G04
                 Remember to process them only in network mode
                 
 error in state - epoch out of range 

 error in state - epoch out of range 

 error in state - epoch out of range 

 error in state - epoch out of range 

 error in state - epoch out of range 
_**(hundreds time same error)**_
.
.
.
**   Correct for time desync
        Correcting carrier phases for a dt drift estimated from desync interpolation
        Correcting pseudo-ranges for a dt drift estimated from desync interpolation
         - 219 code observations marked as outlier
   **   Removing arcs shorter than 11 epochs
         - 139 code observations have been removed
         - 413 phase observations have been removed
   **   Sat Ephemerids: switching to antenna phase center
   **   Loading antennas phase center offset
        Reading antenna PCV/PCO of G constellation/s
   **   Computing position and clock errors using a code only solution
        Applying satellites Differential Code Biases (DCB)
   **   Adding code group delays
        Applying satellites clock errors and eccentricity dependent relativistic correction
        Improving estimation
        Synthesising code observations
        Synthesising code observations
        Setting up SA system using GC1CC2WI 
        PREPRO s0 = NaN (first estimation)
        PREPRO s0 = NaN (iteration skipped)
        Final estimation sigma0 0.000 m
 [ WW ] Warning: Code solution have not been computed, something is wrong in the current dataset

[goGPS_run_20191120_103304.log](https://github.com/goGPS-Project/goGPS_MATLAB/files/3867887/goGPS_run_20191120_103304.log)

clicat, 2019-11-20:
Aaaaah I found your problem is in the "jplephem" not written  by us. I'll search for a solution in the morning, and I'll write you back!

clicat, 2019-11-20:
I actually don't know why that part of the code is not working for you. We always just used it.
Anyhow, I can _maybe_ have a solution.
The broken function is in "jplephem.m" it's a function written for computing sun and moon position giving the JPL ephemeris and it is called from Core_Sky.computeSunMoonPos.
The function reads the values it needs from the file "de436.bin" you can find a copy of it in:

Windows:
< HOME >\AppData\Local\goGPS\
LINUX:
< HOME >/.goGPS/
macOS:
< HOME >/Library/Application Support/goGPS/
where < HOME > is your home folder

while we store the original file in "goGPS_MATLAB/data/reference/jpl".
Delete all the occurrences of it and try to force goGPS to recompute it on your machine.
To do so you'll need to download also ascp01950.436 and other files, you can find them here:
[Issue 63](https://github.com/goGPS-Project/goGPS_MATLAB/issues/63)

What MATLAB version are you running?

Rerun goGPS in a just-opened MATLAB and try to see if it works.

JKeast0, 2019-11-20:
I solved that problem now. The file in here < HOME >\AppData\Local\goGPS was not fully downloaded. I copied the originial file to there and it fixed. But now i'm having another error in NET processing:

> 
>    **   Executing: NET T1:4 R1:3 --IONO
> ----------------------------------------------------------------------------
>         Network solution performing 4 loops of outlier detection on the residuals
>         Estimating G12 widelane using tracking C on frequency 1 and tracking W on frequency 2
>         Setting up SA system using GL1CL2WI 
>         Setting up SA system using GL1CL2WI 
>         Setting up SA system using GL1CL2WI 
>         Setting up SA system using GL1CL2WI 
>  [ !! ] Error: Command_Interpreter - Network solution failed:Undefined function 'LAMBDA' for input arguments of type 'double'.
> ---------------------------------------------------------------------
>  MESSAGE: Undefined function 'LAMBDA' for input arguments of type 'double'.
> 

Btw the version of MATLAB i'm using is R2018a.

ege010, 2019-11-20:
Hi, have you already downloaded the LAMBDA MATLAB code? https://www.tudelft.nl/en/ceg/about-faculty/departments/geoscience-remote-sensing/research/lambda/lambda/

It should be put in goGPS\positioning\lambda\lambda_v3

JKeast0, 2019-11-21:
Yes I downloaded and put in the lambda_v3 folder. Now its giving this:

```
        Network solution performing 4 loops of outlier detection on the residuals
        Estimating G12 widelane using tracking C on frequency 1 and tracking W on frequency 2
 [ !! ] Error: Command_Interpreter - Network solution failed:Index exceeds array bounds.

---------------------------------------------------------------------
 MESSAGE: Index exceeds array bounds.
---------------------------------------------------------------------
```

JKeast0, 2019-11-21:
It worked when turned off these 2 options in pre-processing tab:
Trackings combinations.
Sat clock-re alignment.

Also how can i get position residuals after processing.

giulio-t, 2019-11-21:
Hi, regarding your previous error. Ambiguity fixing in the presence of ionospheric delay is only possible with the new uncombined engine that we are integrating. To remove the previous error you just have to deselect ambiguity fixing from the processing tab. Then the phase float ionofree residual can be found in core.rec(1).work.sat.res.  The rows of the stored matrix are the epochs of the observation( to see them type  core.rec(1).work.time.toString)
and the columns are organized by go_id (an internal sequential index for satellite for gps it is just the prn).

JKeast0, 2019-11-21:
Thanks for all your helps. :)

closed 2019-12-17 by clicat