goGPS-Project/goGPS_MATLAB #72: How to Run Multi-Constellation and Multi-Frequency DD

loveforeverLi, 2019-12-14:
hello everyone,

I run goGPS (beta 1 version) with GPS L1 mode successfully, but when I add more constellations and/or more frequency, some errors like the following occurs:

error RINEX_get_obs (line 300)
        if (~obs_struct.S2(index))

error load_RINEX_obs (line 200)
            [cur_line, obs] = RINEX_get_obs(file_buf, cur_line, num_sat, sat, sat_types, obsColumns, nObsTypes, cc,
            active_sys, first_id_sys);

error goGPS (line 840)
                        load_RINEX_obs(filename_obs, state.getConstellationCollector(), processing_interval);

error goGPSgo (line 68)
    goGPS

Specifically, the error occurs here (RINEX_get_obs, line 300):
        if (~obs_struct.S2(index))
            obs_struct.S2(index) = obs_tmp(index, 2);
        end
index is 40 when I add BDS B1 signal, but the size of S2 only 38.

I'm trying both RINEX 3.0x and 2.1x, the same error occured.

by checking the index can solve the problem:
if(index <= length(obs_struct.S2) && ~obs_struct.S2(index))

but some other problems occurs...

I'd like to know should I debug the RINEX reading process to correct this problem or is this some kown bugs for this version and I just need to change the version? 

Thanks very much!

clicat, 2019-12-17:
Hello @loveforeverLi it seems to me that you are runnig the old 0.5.x version of goGPS. To answer you I need to recap a bit of goGPS story.

In 2017 due to a changed focus in my research I started to rewrite parts of goGPS to make it faster and object oriented. Together with Giulio and Eugenio (that first implemented a new LS engine) we pushed the development in a new direction, we decided to abandon the Kalman Filter approach in favor of the faster (in MATLAB) multi-Epoch joint Least-Squares solver. Today the new goGPS 1.0 beta is a totally new software but it is focused on the processing of quasi static receivers (permanent stations), the older version is still in the repository but no longer maintained. 
We are still updating the code almost daily but we are close to a release candidate, this means that the software is still a bit unstable but mostly usable. We still miss a proper documentation that will follow the first 1.0 final release but other people around the World are already used it.

Just to answer you: you are free to debug the old code but we are no longer dedicating time to it. If you want to try the newer release I suggest you to regularly update your copy from GitHub, but remember that it works only on quasi-static receivers!

Sorry If I can't be more helpful!

clicat, 2020-01-08:
If you are still interested I created the wikinof the project. 
Since this is jot an issue that can be reproduced I’m closing it

closed 2020-01-08 by clicat