goGPS-Project/goGPS_MATLAB #76: tropospheric product 5minutes

syachroel, 2020-01-30:
Dear goGPS team

Congratulations, goGPS is getting better.

But I have a question if I want to set the trop. output. only 5 minutes is there a choice menu?

Because so far the trop. product, the default setting is 30 seconds, but the output that comes out, sometimes starts 29-60-89 .... so it's not consistent if I extract from the data 30 seconds to 5 minutes.

Thanks
Regard
Syachroel

clicat, 2020-01-30:
Thanks @syachroel, by default goGPS computes epoch by epoch parameters, even when splines are selected; in that case, a reduced number of splines is computed and then the synthesized expansion is saved into the receiver workspace.
In a future, we may change this behavior but at the moment, to export in the output object at 5 minutes rate, and then use the export commands to produce output you can use a trick.

Let me explain... at the moment goGPS uses the rec.work object to store the session by session processing data and results, and the rec.out object to store the final multi-session results. I made mandatory the use of the command `PUSHOUT T*` to export the data from work to out, so if you execute:

```
PUSHOUT T* @300s
EMPTYWORK T*
EXPORT T* TRP_MAT
```

your final results will be stored at 5 minutes rate. Note that if for some reason the `PUSHOUT` fails and some values are still present in the work object you risk exporting those data at 30 seconds rate, for this reason after executing the PUSHOUT command I usually perform `EMPTYWORK` (at this point work data are no more needed and I also free up some memory).

clicat, 2020-01-30:
Regarding the times 29-60-89, probably there are some problems with the roundings of the results. I will try to look into it as soon as I have time ;-)

syachroel, 2020-02-04:
sorry still not working by PUSHOUT T* @300s

clicat, 2020-02-04:
mmm write here the exact list of commands you are using

syachroel, 2020-02-06:
% PUSHOUT T* @300s - ERROR: CMD UNKNOWN

syachroel, 2020-02-06:
I'm writing below...
insert here the goGPS command list :

clicat, 2020-02-06:
First it seems you are inserting the command in the inspector, and not in the main command list that you used to execute the processing. This is ok if you have processed just one session and you do not have emptied the receiver workspace. From the inspector you can execute goGPS commands manually. Apparently the inspector does not recognize the command. My guess: have you updated the code? If the instance of MATLAB have not been closed and reopened you need to use the command `clearBranch` in the MATLAB command Window to force cleaning and update!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

clicat, 2020-02-06:
(The rate parameter for the pushout command is a pretty recent update!)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

clicat, 2020-02-19:
The output rate for tropospheric parameters have been added in GUI in the new tab Output

closed 2020-02-19 by clicat