goGPS-Project/goGPS_MATLAB #77: Dynamic model of Receiver

LIFangchao1, 2020-03-24:
Thank you very much for the open-source of goGPS.
There is a Dynamic model setting at the panel of the previous version of goGPS, but I can't find it at the current version. I am not sure it's my low-class error or you didn't set it. 
Also, some questions about functions. In the past few days, I have made a glance of goGPS, There is some difficulty for me to understand the code for single-point positioning for codeStaticPositioning function, such as Astack2Nstack() and solve() function in ls class. I would appreciate it if you can provide any information about how to realize and solve equations for the goGPS.
Thanks for your time.

clicat, 2020-03-24:
Dear @LIFangchao1 thanks for trying goGPS. The new release of goGPS 1.x is just loosely based on the 0.x. We no more use Kalman filter but batch LS solutions, (some more information can be found on the project website [gogps-project.github.io](gogps-project.github.io) and we focused our current effort in creating a software for the processing of static stations, maybe this will change in the future but at the moment there are no current plans to reintroduce dynamic processing, we just don't have enough men power to do it.
The code could be complicated we know, it became bigger and bigger but it seems to me that you are already doing a good job. 
To be honest, I'm not sure I understood your question properly, can you rephrase "realize and solve equations for the goGPS"?
Anyhow, I could try to answer your question but the first author and primary maintainer of the engines is @giulio-t @tagliagiulio, he is doing a great job in the development of the solver that will be part of his PhD thesis and he can probably answer you in a better and more complete way. I can assign to him this "request" but I kindly ask you if you can reformat your question in a set of more detailed points.
Best regards,
Andrea

LIFangchao1, 2020-03-25:
Thanks for your reply and time.
Sorry for my unclear description. It is difficult for me to understand the codeStaticPositioning function, which you called batch LS. Would you like to provide some literature about batch LS?

tagliagiulio, 2020-03-27:
Hi LIFangchao1. 
Which part of the code you have problem to understand?
Batch LS are simply Weighted Least Squares.  https://en.wikipedia.org/wiki/Weighted_least_squares
The term batch is used in the sense that observations from all epochs are compensated together.

LIFangchao1, 2020-03-27:
> Thank you. All of the questions has been solved, and I have complicated this question before.

tagliagiulio, 2020-04-02:
the asking user says the question has been answered

closed 2020-04-02 by tagliagiulio