goGPS-Project/goGPS_MATLAB #79: New mapping function

pyuan111, 2020-06-02:
Dear developers, is it possible to add the newly released VMF3 (1*1 degree) mapping function into goGPS?

clicat, 2020-06-13:
Yes, it is. it requires us to have a little bit of spare time. We want to add many more mapping functions but modifying the interface and set up the automatic download of more resources is not  so immediate. The actual main developers of goGPS are @giulio-t and me, but if anyone wants to do something we are always happy to help, unfortunately, it is not so easy to find enough time to do everything we have planned. I'm keeping the request open, I hope that soon we will find time to add it.

clicat, 2020-09-01:
Done by @giulio-t to be tested better but it is there!

closed 2020-09-01 by clicat