goGPS-Project/goGPS_MATLAB #81: GUI interface problem

Adabapt, 2020-06-19:
Hi developers,

I have some problems with the GUI interface. In fact, when I launch goGPS, it opens a default PPP project but I don't have the interface to create a new project or to load an others project, and in the matlab console I have these errors : 

Index exceeds the number of array elements (0).

Error in GUI_Edit_Settings/updateResourcePopUpsState (line 2755)
            this.rpop_up.Parent.Children(2).String = sprintf('Supported satellites: "%s"', center_ss{value});

Error in GUI_Edit_Settings/updateUI (line 2996)
                this.updateResourcePopUpsState();

Error in GUI_Edit_Settings/openGUI (line 397)
            this.updateUI();

Error in GUI_Edit_Settings (line 94)
            this.openGUI(flag_wait);

Error in GUI_Edit_Settings.getInstance (line 108)
                this = GUI_Edit_Settings(flag_wait);

Error in Core_UI/openGUI (line 778)
            this.main = GUI_Edit_Settings.getInstance(flag_wait);

Error in goGPS (line 128)
        ui.openGUI(flag_wait);

my interface when I launch goGPS :
[image]

clicat, 2020-06-24:
Hi @Adabapt,
 sorry for the late answer, I tried to clone the project in a new folder and execute goGPS as you can download it. 
The GUI works without any problem in my MATLAB 2018a and MATLAB2020a, what version are you using? 
In your screenshot, I cannot see the menubar of the goGPS window. where the menu for the guided creation of a project is. At the bottom of the window however, there is the load button to load other projects. You can always modify the current and save it in another config file (it's a text file you can also edit it manually).

I cannot reproduce your error, I can just suggest updating your code to the latest commit, executing clearBranch and then relaunch goGPS.

Adabapt, 2020-06-24:
I run GoGPS in MATLAB2020a.
So I update my code with the last commit but I have an error when I launch the clearBranch script 

Unable to resolve the name GUI_Edit_Settings.closeGUI.
Error in clearBranch (line 3)
GUI_Edit_Settings.closeGUI

And when I launch goGPS the menubar don't appear and I can't choose the type of the project or create an other one.
I know I can change some arguments in the settings file but I can't change the type of the project.

clicat, 2020-06-24:
It seems you don't have the code added correctly to the path
Move to the goGPS folder and type in MATLAB command line
```
>> addPathGoGPS
>> clearBranch
```
goGPS have been always run by us from the goGPS folder.
then the command `addPathGoGPS` add to the MATLAB path all the folders needed by goGPS
It seems that you executed `clearBranch` with not the right path added otherwise it should find the `GUI_Edit_Settings.closeGUI()` function.
The command `clearBranch` is used by us to clean all the singletons object when we change something in the Classes code, if you run a fresh MATLAB is not needed.

The error that you first posted is on a line that doesn't match with the current code:

_Error in GUI_Edit_Settings/updateResourcePopUpsState (line 2755)
this.rpop_up.Parent.Children(2).String = sprintf('Supported satellites: "%s"', center_ss{value});_

[image]

The function is in another line and at 2755 I have an empty line.

Could you repost a full screenshot of your screen with the actual error in the command line (executing the most recent code)?

Adabapt, 2020-06-24:
After the execution of addPathGoGPS, the clearBranch script stop to produce some errors but my first problem with the GUI wasn't fix and I continue to not have any menubar when I launch goGPS.
I have also a little issue with the goGPS log window which clean randomly.

[image]

clicat, 2020-06-24:
I managed to reproduce the bug and fix it (I hope). It was due to a clf command on an axis in the figure. It shouldn't remove the menus but it does (on MATLAB 2020a).

Check out the last commit f8f567f that I just pushed on the branch.

Let me know if the bug is fixed also for you!

clicat, 2020-06-24:
About the log window being cleared, it is due to bad synchronization of the interface, if you click on a window before MATLAB finishes some updates it may happen.

Adabapt, 2020-06-24:
About the log window I noticed it was about the update of the window, It was just to let you know.
And for the GUI interface it's completly fixed.
I can create new project with the menubar or change the type of the current project.

[image]
[image]

Thanks for your time (also during the night).
Can I close the issue ?

clicat, 2020-06-25:
Perfect

closed 2020-06-25 by clicat