goGPS-Project/goGPS_MATLAB #82: try other MET files formats beside  meteorological RINEX met files

Eb2010, 2020-07-26:
the stations that I processed are not IGS stations, and I have csv files for the meteorological parameters, is there anyway I could input these files rather than  meteorological RINEX fiels

clicat, 2020-07-27:
Is it possible, but it requires to read your format, do you have an example? Is it standard? Do you have the coordinates of your meteo station?

You can convert your data in Meteorological RINEX format, or we can try to build a reader for your data.

Eb2010, 2020-07-27:
[en_climate_hourly_NU_2401203_01-2019_P1H.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/4981716/en_climate_hourly_NU_2401203_01-2019_P1H.zip)

this is one of my files, I want to ask, is it possible to convert my file to Meteorological RINEX format?? how can I make that?

clicat, 2020-07-27:
I create a simple importer (not very optimized) of your CSV, you can find it in Meto_Data.m (please update your code to the latest commit)
Running the following example you will be able to save a set of METEOROLOGICAL RINEX like the ones in the attachment

[Eureka_2019.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/4982791/Eureka_2019.zip)

Example usage:
```
md = Meteo_Data.importCanadianCSV('./path_to_your_csv/en_climate_hourly_NU_2401203_01-2019_P1H.csv', 0);
md.export('./path_to_your_met_folder/')
```

I made a simplistic hypothesis, there is only one station per CSV.
Using met RINEX file is not always very fast, but it is the only standard that I found.
By the way the second parameter of Meteo_Data.importCanadianCSV is the hortometric height of the meteorological station (unknown in your data, I set it to ZERO).

Try and let me know, in our examples it works!

Eb2010, 2020-07-27:
should I run it, separately in the command window of matlab? 
when I run this commands in the command window before running goGPS 
I get this error message

[ WW ] Warning: The settings field min_p_epoch is not valid => using default [ 60 ]
 [ ok ] File "C:\goGPS_MATLAB-goGPS_1.0_beta\data\project\default_PPP\Config\settings.ini" found, settings imported!
 Exporting met data to C:\goGPS_MATLAB-goGPS_1.0_beta\data\project\default_PPP\station\MET\
 [ !! ] Error: Export failed - Invalid file identifier. Use fopen to generate a valid file identifier.
 Exporting met data to C:\goGPS_MATLAB-goGPS_1.0_beta\data\project\default_PPP\station\MET\
 [ !! ] Error: Export failed - Invalid file identifier. Use fopen to generate a valid file identifier.
 Exporting met data to C:\goGPS_MATLAB-goGPS_1.0_beta\data\project\default_PPP\station\MET\
 [ !! ] Error: Export failed - Invalid file identifier. Use fopen to generate a valid file identifier.

is there something wrong with me??

clicat, 2020-07-27:
You can ignore the first warning (we added a new field) 
While you just have to pass an existing valid folder to the command export (put a slash at the end of the path to be sure)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Eb2010, 2020-07-27:
Done, really thanks so much for your help

closed 2020-07-27 by Eb2010