goGPS-Project/goGPS_MATLAB #83: ERROR: Receiver 1: eurc has not been pre-processed, PPP is not possible.

Eb2010, 2020-07-29:
Dear Dr.
please I have two error messages while processing RINEX files, and I don't know what was wrong I did?
 
ERROR: Receiver 1: eurc has not been pre-processed, PPP is not possible
ERROR: The requested atmospheric parameter have not been computed or stored

any help please?

clicat, 2020-07-31:
That file is particularly bad, the code solution was failing. Try with the latest update. It should work now

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Eb2010, 2020-07-31:
thanks, I will try now and let you know

clicat, 2020-09-04:
No more news, I declare the issue closed...

Eb2010, 2020-09-04:
Hi,
Unfortunately, the program doesn't work with me till now, although I processed different stations
so, thanks for informing me

GABR, Ebtesam
M.Sc Student / Teaching Assistant
Radiophysics laboratory
Physics Department, Faculty of Science
University of New Brunswick
Fredericton, NB. E3B 3V6
T: (506) 4407280
E: efarid@unb.ca

⚠ External message: Use caution.

clicat, 2020-09-04:
Ok I reopen the bug and I'll test it again, I remember that I was able to get a solution. Did you update the code? Do you receive the same error?

Eb2010, 2020-09-04:
yes, I update the code
I'll update it again 

do you have any idea, what's the problem, why I have such an error? or what should I do to fix this problem

clicat, 2020-09-04:
I just tried to reprocess EURC0010.18o it's not the best dataset ever but it works. Please attach the full transcript of your error. If you can attach your config file so I can try to replicate your error on my computer!

Eb2010, 2020-09-07:
I can't attach the config file, so I take it a copy and paste here

#----------------------------------------------------------------------------------
[SOFTWARE]
#----------------------------------------------------------------------------------
# goGPS config file
version = "1.0b7"

#----------------------------------------------------------------------------------
[PARALLELISM]
#----------------------------------------------------------------------------------

# Location of the communication dir for parallel message passing interface
# Absolute or relative to execution path
com_dir = "com"

#----------------------------------------------------------------------------------
[PROJECT]
#----------------------------------------------------------------------------------
# Name of the project
prj_name = "Default PPP project"
# Home of the project
prj_home = "../data/project/default_PPP"

# NOTES ON THE NAMING CONVENTIONS:
# Special Keywords that can be used in file names:
#  - ${WWWW}    4 char GPS week
#  - ${WWWWD} 4+1 char GPS week + day of the week
#  - ${D}       1 char day of the week
#  - ${3H}      2 char GPS hour (00, 03, 06, 09, 12, 15, 18, 21)
#  - ${6H}      2 char GPS hour (00, 06, 12, 18)
#  - ${HH}      2 char GPS hour
#  - ${QQ}      2 char GPS quarter of hour (00, 15, 30, 45)
#  - ${5M}      2 char GPS five minutes (05, 10, ... , 55)
#  - ${YYDOY} 2+3 char GPS year + day of year
#  - ${YYYY}    4 char GPS year
#  - ${YY}      2 char GPS year
#  - ${MM}      2 char GPS month
#  - ${DD}      2 char GPS day
#  - ${DOY}     3 char GPS day of the year
#  - ${S}       1 char session

#----------------------------------------------------------------------------------
[SESSION]
#----------------------------------------------------------------------------------
# "sss_" parameters define the session of observation, they are used
#        to substitute special keywords in file names
# Working session - first data of observation to consider (yyyy-mm-dd <HH:MM:SS>)
# mainly used to detect the name of the file to process
sss_date_start = "2017-04-08 00:00:00"
# Working session - last data of observation to consider (yyyy-mm-dd <HH:MM:SS>)
sss_date_stop = "2017-04-09 23:59:59"
# Id character sequence to be use for the session $(S) special keyword
# (e.g. "01233456789ABCabc")
sss_id_list = "0"
# First session id (char of sss_id_list)
sss_id_start = "0"
# Last session id (char of sss_id_list)
sss_id_stop = "0"
# Flag to base the sessions on the RINEX files
sss_file_based = 0
# Session duration in seconds
sss_duration = 86400
# Session buffer in second [left right]
sss_buffer = [10800 10800]

# Computing the troposphere on multiple sessions (even with buffering)
# could produce discontinuous series, at the change of session.
# To produce a smooth solution, the session from the past can be 
# connected to the new one weightning the two buffered areas.
[... 952 lines cut ...]
# 
#              Admissible session parameters: S, T
# 
# ----------------------------------------------------------------------------------------------
#  - PAR       Parallel section start (run on targets)
#              use T$ as target in this section
# 
#              Admissible session parameters: S, T, P
# 
# ----------------------------------------------------------------------------------------------
#  - END       For loop end or Parallel section end
# 
# ----------------------------------------------------------------------------------------------
# 
cmd_001 = "FOR S*"
cmd_002 = "FOR T*"
cmd_003 = "LOAD T$"
cmd_004 = "REMSAT T$ G08"
cmd_005 = "PREPRO T$"
cmd_006 = "PPP T$"
cmd_007 = "END"
cmd_008 = "PUSHOUT T*"
cmd_009 = "END"
cmd_010 = "SHOW T* ZWD"
cmd_011 = "EXPORT T* TRP_SNX"

closed 2020-09-07 by Eb2010