goGPS-Project/goGPS_MATLAB #84: Wiki for sub-tabs of Tab-Processing is missing?

pyuan111, 2020-09-07:
Dear goGPS developers,

I am trying to use goGPS for my study on water vapor.
However, it seems that the wiki of the sub-tabs of the Tab Processing is missing.

Clicking the links shows that they are not found:
https://github.com/goGPS-Project/goGPS_MATLAB/ProcessingTab/#tab-data-selection
https://github.com/goGPS-Project/goGPS_MATLAB/ProcessingTab/#tab-atmosphere
https://github.com/goGPS-Project/goGPS_MATLAB/ProcessingTab/#tab-pre-processing
http://github.com/goGPS-Project/goGPS_MATLAB/ProcessingTab/#tab-generic-options
https://github.com/goGPS-Project/goGPS_MATLAB/ProcessingTab/#tab-ppp-parameters
https://github.com/goGPS-Project/goGPS_MATLAB/ProcessingTab/#tab-network-parameters

Could you please add the information to the wiki?

Best regards,
Peng

clicat, 2020-09-07:
Hi Peng,
     the wiki is a mirror of the documentation present on the website. 
I always suggest to use the website to read documentation, there the links are working:
[https://gogps-project.github.io/wiki/ProcessingTab/#tab-data-selection](https://gogps-project.github.io/wiki/ProcessingTab/#tab-data-selection)

A few newer settings might be missing, we will update the wiki when other planned modifications are finished.
(e.g. added VMF3, more ionex centers....)

closed 2020-09-07 by clicat

pyuan111, 2020-09-07:
Dear Gatti,
Thanks a lot for your reply.
Best regards,
Peng