goGPS-Project/goGPS_MATLAB #85: is there any function to get velocity and time as an output

JasmienHassanien, 2020-09-08:
hi, i need a help,
I'm using goGPS V0.4.3
is there any way to check the accuracy of my receiver in terms of  velocity and time , the output is considering the position only. i checked the (doppler_shift_approx.m) function for velocity, and (transmission_time.m) for time

also i wonder from where can i adjust the number of epoch to be processed in that version.
thank you.

ege010, 2020-09-08:
Hi Jasmien, unfortunately that version is quite old and no longer supported.

JasmienHassanien, 2020-09-08:
actually I am new to goGPS and hoping to get some help. I couldn't find any publication or user guide for the new version that's why i started with older one, any recommendation or suggestion to get some guide?
thank you.

ege010, 2020-09-08:
Sure, you can have a look at the documentation here: https://gogps-project.github.io/
Closing this issue now. If you have other specific questions on the new version, please feel free to open a dedicated issue for each one.
Thanks!

closed 2020-09-08 by ege010

fatenReda, 2022-02-23:
hi sir ,
it is my first time to deal with goGps program.
is this program processing real time kinematic data?
what is the latest version of this program?