goGPS-Project/goGPS_MATLAB #89: unable to run goGPS_v0.4.3 in matlab2018b

LakshmiCBR, 2020-10-03:
goGPS GUI initialization completed in 30.70 seconds
Error using exist
The first input to exist must be a string scalar or character vector.

Error in goGUIclass/updateLEDstate (line 2168)
if exist([data_path file_name],'file')

Error in goGUIclass/checkUIdependencies (line 1914)
obj.updateLEDstate();

Error in goGUIclass/updateGUI (line 1369)
obj.checkUIdependencies();

Error in goGUIclass/browseINIFile (line 2338)
obj.updateGUI();

Error in goGUIclass/syncFromGUI (line 2035)
obj.browseINIFile();

Error in gui_goGPS>bINI_Callback (line 359)
goGUI.syncFromGUI(goGUI.idUI.bINI);

Error in gui_mainfcn (line 95)
feval(varargin{:});

Error in gui_goGPS (line 64)
gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)gui_goGPS('bINI_Callback',hObject,eventdata,guidata(hObject))

Error using uiwait (line 81)
Error while evaluating UIControl Callback.

Please helpme running goGPS.....it is failing with the above errors. I tried using clearBranch.m too. But no use

clicat, 2020-10-03:
Dear @LakshmiCBR clearBranch have been created to clear variables and singleton classes of the new goGPS. The very old 0.4.3 version (legacy) is basically a different software with respect to goGPS 1.0. I can support only the latest 1.0 beta version. If you don’t need Kalmann Filter or dynamic processing I suggest you to use the latest version of goGPS.
Sorry,
    Andrea Gatti

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

LakshmiCBR, 2020-10-06:
> Sent with GitHawk <http://githawk.com>

Dear Andrea Gatti,

Thank you very much for your kind reply.
As you suggested I installed latest version V1.08b and trying for a new
project for PPP. I am getting a warning continuously
WARNING: CRD file seems empty or corrupted
 May I know what does it mean and how to rectify the error,please.

clicat, 2020-10-30:
Did you saved the CRD file? Have you tried to open it, does it contains coordinates?

clicat, 2020-12-09:
One month old problem with no answer, I cannot reproduce the bug, thus I'm closing the issue. Reopen it if you can provide additional informations.

closed 2021-01-25 by clicat