goGPS-Project/goGPS_MATLAB #90: network adjstment fail

lbenvenuto91, 2020-10-13:
Hi,

I'm trying to compute a network adjustment but I got an error that I can't understand. The error is:

ERROR:            Command_Interpreter - Network solution failed:Index exceeds the            number of array elements (0).
--
   file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\LS_Manipulator_new.m"   line: 1295   fun: LS_Manipulator_new.removeFullRankDeficency    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\LS_Manipulator_new.m"   line: 1662   fun: LS_Manipulator_new.solve    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\Network.m"   line: 475   fun: Network.adjustNew    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\Command_Interpreter.m"   line: 2067   fun: Command_Interpreter.runNet    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\Command_Interpreter.m"   line: 1425   fun: Command_Interpreter.exec    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\Command_Interpreter.m"   line: 1305   fun: Command_Interpreter.exec    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\Core.m"   line: 1129   fun: Core.exec    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\obj\Core.m"   line: 1114   fun: Core.go    file: "C:\Users\gterg\Documents\GitHub\goGPS_MATLAB\goGPS\goGPS.m"   line: 182   fun: goGPS

The PREPRO phase seems to be ok, but when it starts the net computation I get this error.

I attach here the configuration file:
[config1.txt](https://github.com/goGPS-Project/goGPS_MATLAB/files/5369850/config1.txt)

I'm running the last version of goGPS (V1.0b8)

clicat, 2020-10-26:
Hi, thanks for the report of the bug, I’m not directly working on this part of the processing, but I know that it is under revision. I hope that the bug will automatically disappears with future updates.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

clicat, 2020-12-09:
Hello @lbenvenuto91, I just looked out of curiosity to your config file, it seems you were trying to process a full network of 18 stations @30 seconds for 30 hours. The open version of goGPS cannot process such an amount of stations unless you have a machine with a looooooot of RAM and CPU power. Maybe this was the cause of the error. 
Sorry I'm closing the bug as it is impossible for us to solve the problem with the open release of goGPS.

closed 2020-12-09 by clicat