goGPS-Project/goGPS_MATLAB #91: troubling with understanding this program..

gpc-2020, 2020-10-17:
My name is Heeng from South Korea.I am university student in Korea
I found your gogps project and tested it for studying purposes. However the program said there is or are error/errors with what I wanted to test (attachment).

I have collected RINEX observation data and navigation data from mobile.
When I inserted observation data with GPS, Galileo, Beidu, GLONASS etcthe program stopped as mentioned above. 
How can I solve this problem?
RINEX data was collected on 2020.08.31 in Korea.
 thus time would be +9 from observation data file time. 
the purpose was to find positioning of x,y,z epoch per second.
also as my poor English i do have trouble with using the program. 
and  I could not 100% understand how to use the program. 
Thank you
[New_Project.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/5394964/New_Project.zip)
[image]

Heeng

clicat, 2020-10-19:
Sorry Heeng I’m on holidays so I’ll be short. There is a tab called resources, there you need to select the provider of the orbits that you want to use. IGS default final orbits only support GPS and GLONASS. If you want all the constellations you can choose orbits from something like CODE MGEX

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

gpc-2020, 2020-10-26:
> Sent with GitHawk <http://githawk.com>

Hello Andrea

I hope you are having amazing holidays.

I did what you suggested then new program popped out
However, when i clicked every buttons on every Plots (1 to export)
about 90% of them did not work with comment of command not found or
something similar to it
i would like to get coordinates such as ecef coordinate xyz etc
what have i missed to get them?

thank you

Heeung

clicat, 2020-10-26:
The software can create plots depending on the kind of processing you started, and on the results that you kept or exported. Some of the plots works only if you have processed multi-session. Unless in the last week the program was broken, I believe that 99% of the “non working plots” are not working because of your processing strategy. Note that goGPS at the moment estimates one position per session, if you processed one session only, no plots on the stability of the coordinates are possible.
If you find a bug I can try to help, but if you say 90% of the plots don’t work I can’t. I don’t want to seem rude, If you ask me one single specific question I can answer but if you ask me explain how gnss processing works I definitely have not enough free time to do it.

Moreover last warning: mobile data are of such bad quality that to process them you need to really understand what is going on, and be able to fix regularization and processing parameters...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

gpc-2020, 2020-10-30:
Dear Andrea

I hope you are well with the crisis that is happening around the world.

I re-read github website then tried more tests however that still did not
work.
thus i made explanation with screenshots of the problems im facing.
i hope that is better for you to see the causes.
i have attached 2 trials i did: one with just one data and another one with
four datas.

it would be wonderful if you can read through the attachment and guide me
what I did wrong.

thank you
Heeung
 Test1.zip
<https://drive.google.com/file/d/1UAd5YmI_Pkg6HYwuuJe9zgfJRovu5XFW/view?usp=drive_web>
 Test2.zip
<https://drive.google.com/file/d/1JLdU6hvYbxuCn9FH4zSZYt-uF0s32RGV/view?usp=drive_web>

clicat, 2020-10-30:
Hi,
I requested you access to the google drive files, it seems that they cannot be downloaded without authorization

clicat, 2020-10-30:
Hi, I checked your files, Test 1 was full of mistakes and wrong configurations, I sent you a reviewed document.
Anyhow as previously stated processing cellphone RINEX requires a higher degree of comprehension of the adjustment and a lot of effort in calibrating parameters. Sorry, but I have no time to help you process the data. If you find any bug, with higher quality data you are free to post it here. Maybe we have time to solve it. 

Cordially,
Andrea

P.s. I'm closing the bug, because it's not a bug and cellphone RINEX files are not really supported

closed 2020-10-30 by clicat

gpc-2020, 2020-10-30:
Thank you for your time and wonderful helps.

And Test2.zip had same mistakes and wrong configurations, am i correct?
(The one with 4 datas)

Heeung