goGPS-Project/goGPS_MATLAB #92: Clock and iono file download problem

maurob89, 2020-10-29:
Good evening, I am Mauro Burgassi, university student and I have tried your software to make the p.p.p. of a single frequency receiver. 
During the processing the software produces some errors and can't download the clock and iono files. 
The software is updated to the latest version. The anomalous thing is that on a different pc it works without any problem. 
I enclose "config.ini" and "log" files.
Thank you very much in advance 
[Config e log.zip](https://github.com/goGPS-Project/goGPS_MATLAB/files/5460151/Config.e.log.zip)

clicat, 2020-10-30:
Ciao Mauro,
from what I see from the log it seems that your computer (or MATLAB and aria2c) cannot connect to the FTP. It looks more like an internet access permission issue. Do you have a firewall or are you in a university that could block these addresses?
Try to manually access FTP, can you?
I say so because from the log the software cannot access any remote location, and this is very strange unless there's a network problem (and this is even more true if you say that from another computer it works).
Secondly, just a warning: the PPP of a single frequency receiver, without interpolating its ionospheric error from close dual-frequency receivers might be almost unuseful.
;-) 
Out of curiosity, at which university are you studying?

clicat, 2020-11-03:
@maurob89 have you solved your problem?

maurob89, 2020-11-04:
Good evening, unfortunately I was unable to solve the problem. I tried to temporarily disable the antivirus protection but without success, while direct access via the web to the site is without problems. I also tried to install the software on another pc but still the same problem. Could it be that the antivirus common to both PCs does not allow authorization anyway? I work on Windows 10 and with McAfee antivirus software. Thanks

kazimz, 2020-11-19:
Hi
I guess this is the reason (see attached image) that we are unable to download Clock and iono files?
[image]

clicat, 2020-12-09:
Sorry I cannot reproduce the problem, you can always put manually the resources you need in the proper folders, and goGPS will work with what it has.

closed 2020-12-09 by clicat