goGPS-Project/goGPS_MATLAB #96: error running goGPS

Eb2010, 2020-12-20:
Dear developers,

I have this error message while I run (goGPS.m)

{
Error using mkdir
The system cannot find the path specified.

Error in Core/initLocalPath (line 803)
                mkdir(this.local_storage)

Error in Core/init (line 751)
            this.initLocalPath();

Error in Core.getInstance (line 199)
                                this.init(force_clean);

Error in goGPS (line 114)
            core = Core.getInstance(true, false); % Init Core

}

I am using R2020b version of Matlab.

any suggestions/help, please?

clicat, 2020-12-20:
Hi, the error is a bit odd, that code always worked, in that function, the system tries to create a local storage folder to store some orbital parameters.
Depending on the operative system should be in different position:
```
- WIN: [home '\AppData\Local\goGPS']
- LIN: [home '/.goGPS']
- MAC: [home '/Library/Application Support/goGPS']
```
The home is get from environmental variable `HOME` on Unix, and `[getenv('HOMEDRIVE') getenv('HOMEPATH')]` on Windows. 
Try to check your permission to that folder, or try to create it by yourself.

Just in case, I'm pushing a modification to the code to add more verbosity to fix your problem,
Pull the latest commit from git and retry, please.

Eb2010, 2020-12-20:
pull the latest one.
it works now well.
Thanks

Eb2010, 2020-12-20:
I am so sorry, but I removed the folder and re-clone it again to remove any change that could happen to the (data) folder.
why it gives me an error 
{
"C:Error: Creating "H:\\AppData\Local\goGPS" seems impossible, check your permissions!
}

why it tries to access H:/ 
I don't have an H partition, also I set up the software on C:/, not any other partitions.

clicat, 2020-12-21:
I don't have any computer running Windows to test it, but it seems that your copy of Windows have some environmental variables that are misconfigured, or that cannot be read by MATLAB. What happen if you type in the command window of MATLAB the following command:
```
[getenv('HOMEDRIVE') getenv('HOMEPATH')]
```
???

If you want a fix just for your computer, you can modify the code of goGPS:
in Core.m line 792, set the `home` variable to point to your home folder (instead of `home = [getenv('HOMEDRIVE') getenv('HOMEPATH')];`).

closed 2020-12-22 by Eb2010