goGPS-Project/goGPS_MATLAB #97: Why don't GUI programs use. mlapp?

kedoneyes, 2020-12-24:
Is there a plan to separate the GUI program from the handler? Now the error correction code cannot be used alone.

clicat, 2021-01-25:
Sorry, I did not see the notification of this issue, I'm reading it now. What do you mean by "error correction code"? 
I'm using goGPS daily both with GUI enabled and disabled, but I have some additional internal code that maybe have some bugs fixed with respect to the open release. If you can be more specific I can try to solve the problem.

The GUI is basically an Editor of the settings (goEditSettings) + an Explorer of the results (goInspector) + a log Visualizer. goGPS was born to be used by command line, GUI should already be separate, the settings are text files exported with a ton of comments, and there should be no need for UI.
When we programmed goGPS GUI we were using older versions of MATLAB without, or with preliminary App Designer so using it was not an option. The GUI_Layout_Toolbox is free, is very well written, and super flexible, so it was the best choice. Anyhow, we are using some non-standard UI element (Java) and sooner or later we will need to rewrite the UI when MATLAB will drop features.

clicat, 2021-02-19:
The new release RC1 has been pushed as the main branch. To reach stability earlier, we decided to drop some features, and command-line execution is one of them. The goal is to have software for student and researchers testing their projects in post-processing. I'm closing this issue since too much time is passed since the latest message.

closed 2021-02-19 by clicat