goGPS-Project/goGPS_MATLAB #98: problem in launching goGPS from crontab

lbenvenuto91, 2021-02-19:
Hi,

I'm using goGPS to calculate ZTD from a small GNSS receivers network. I wrote a python script as I want to calculate ZTD every day automatically. This python script, and goGPS are in an Ubuntu server 18.04. As I manually excecute my script everything is fine and goGPS can finish the elaboration. The problems come when I run the script (and so also goGPS from crontab). I can't actually understand the problem but it seems that aria2c is not able to download all necessary files when it's run from crontab. I don't understand the reason why? Any ideas?

The line in the /etc/crontab file is the following:

34 17 * * * gter /usr/bin/python3 /home/gter/REPOSITORY/concerteaux/concerteaux_gnss/elabora_dati_gnss.py > /tmp/elaborazione_dati_gnss_ce.log 2>&1

The goGPS version is the v 1.0b8 and the Matlab version is the R2020a
In the elabora_dati_gnss.py script, apart from other commands, the command goGPS( inifile',0) is launched.
I attach the logs of the elaborations from crontab and from comand line.
[goGPS_run_ZTD_elaborations_20210217_090008_crontab.log](https://github.com/goGPS-Project/goGPS_MATLAB/files/6011638/goGPS_run_ZTD_elaborations_20210217_090008_crontab.log)
[goGPS_run_ZTD_elaborations_20210219_172932_command_line.log](https://github.com/goGPS-Project/goGPS_MATLAB/files/6011639/goGPS_run_ZTD_elaborations_20210219_172932_command_line.log)

ege010, 2021-02-21:
Replied by email.

closed 2021-02-21 by ege010

rmarzocchi84, 2021-04-02:
If useful the problem is clear.  Running the software from crontab "aria2c" command line application doesn't work properly. 

```
Executing 
   aria2c -c -i ./reserved/tmpAriaDownload.lst -d /home/gter/REPOSITORY/goGPS_MATLAB_git/data/satellite/EPH/2151
   File download list:
        ftp://ftp.aiub.unibe.ch:21/CODE/2021_M/COD21511.EPH_M.Z
        ftp://ftp.aiub.unibe.ch:21/CODE/2021_M/COD21510.EPH_M.Z
        

04/02 17:47:10 [[1;32mNOTICE[0m] Downloading 2 item(s)
gzip: /home/gter/REPOSITORY/goGPS_MATLAB_git/data/satellite/EPH/2151/COD21511.EPH_M.Z: No such file or directory

```

I hope it can be useful for your developement!